repetier/RepetierFirmware #476: Temperature spikes when using Repeteir Firmware and Octoprint

wildcardfox, 2016-02-19:
Hi, let me preference this by stating that I opened an error report with Octoprint and they told me to come here and report it as a possible firmware bug. Below is what I posted to Octoprint. 

What were you doing?

I was starting a print.

What did you expect to happen?

A constant temperature curve that gradually shows the rise, and reporting or current temperature readings when the print finishes.

What happened instead?

Temperature spikes when heating the hotend. It shows a drastic fall for one reading like in the terminal log that I have attached where the temperature reporting goes from 157.61 to 59.49 then to 161.77.

Recv: T:157.61 /220 B:44.89 /43 B@:0 @:255
Recv: T:59.49 /220 B:44.89 /43 B@:0 @:255
Recv: T:161.77 /220 B:44.89 /43 B@:0 @:255

When the print is finished current temperatures fail to report.

The problem occurs every time on both of my machines that are running Repetier firmware 0.92.8. I have checked the thermistors and the readings on Repetier Host, and the temperature curve is constant.

Branch & Commit or Version of OctoPrint

Version 1.2.9

Printer model & used firmware incl. version

Two Delta custom printers (Rostock and Large Kossel running Arduino Due with RADDS shields); Repetier 0.92.8

Browser and Version of Browser, Operating System running Browser

Octoprint, Browser Safari, Google chrome.

Link to octoprint.log

https://gist.github.com/wildcardfox/55cb3ec97d9bd63974bd

Link to contents of terminal tab or serial.log

terminal

http://pastebin.com/pp82KMJ1

serial.log

https://gist.github.com/wildcardfox/9696c09e86ddc1ec24d6

Link to contents of Javascript console in the browser

Screenshot(s) showing the problem:

Screen shot

http://i979.photobucket.com/albums/ae280/wildcardfox/wildcardfox028/Screen%20Shot%202016-02-19%20at%208.34.59%20AM_zpsxov7z8kk.png

repetier, 2016-02-20:
That is something hard to find. I do not think the firmware is reading these wrong temperatures. If you watch it more precisely you see that when it happens it is the first digit that is missing in transmission.

As you said the Host does not show the spikes, which simply means that the data gets send completely here. You should enable logging in host to be sure. Could also be that host just did not interpret the wrong value since the 1 got maybe a different char making it not a number.

Also the question arises what the difference is. Does Octoprint run the same PC as the Host? Or is it a different one like a raspberry pi? What I think is that you have serial communication errors somewhere.  I think firmware sends correctly at the serial output and I think Octoprint receives the wrong lines. I do not know octoprint enought to say if it filters illegal chars, so you do not see them or if it is just a invisible char or completely not there. So my guess is none of the software packages is the reason but the hardware/os drivers and their error corrections between them. The back direction has no higher level error correction like gcode that gets send, so you see the result unfiltered. Maybe not nice but there is no real error that has influence on your print.

You might try different baud rates/other usb cables/active usb hub. These things often have an influence on communication stability.

wildcardfox, 2016-02-20:
Awesome. Thanks for detailed response! The host runs on a MacBook and Octoprint is on a RasberryPi. The Mac is up to date and brand new as well as the software on the Pi's. Also the I've cables are also new. I'll attempt different baudrates but like I mentioned It doesn't happen when I'm using your guys Host, so the end result may be that I just switch to Repetier Server for my wifi printing needs.

repetier, 2016-02-20:
Repetier-Server would be a try for additional testing. If it is the Pi/Linux causing the problem the server will of course show the same error. I had cases where power was not stable enough from pi and with active hub it worked. That may also be the difference to mac which probably delivers more power on the usb.

wildcardfox, 2016-02-21:
Yah thinking that it is power supply related would make sense except for this problem only just showed up. I was running Marlin and Smoothieware on a Rambo and a Azteeg X5 V2 before switching to the Due and the Raspberry Pi and Octoprint were perfect, so makes me kinda rule out a power supply that is not sufficient. I'll try the server and see what happens. Hopefully the issue is only occurs with Octoprint.

wildcardfox, 2016-02-21:
Maybe you're right. Perhaps the Arduino Due's take more power Than the raspberry Lu can deliver. I'll switch over to some powered use hubs and see what happens. Thanks for he advise.

rcarlyle, 2016-02-21:
I'm seeing the same behavior with OctoPi and Due/RADDs on a beefy PSU. The 100's digit of the temp report gets dropped intermittently while preheating. I have seen it loose 1s and 2s (ie 217C becomes 17C). It will often happen 2-3 times while preheating at the start of a print but doesn't seem to happen otherwise. 

Example:
Recv: T:213.33 /220 B:60.00 /60 B@:34 @:161
Recv: T:213.86 /220 B:60.00 /60 B@:13 @:160
Recv: T:214.39 /220 B:60.00 /60 B@:34 @:159
Recv: T:214.91 /220 B:60.00 /60 B@:13 @:158
Recv: T:215.44 /220 B:60.00 /60 B@:34 @:156
Recv: T:15.96 /220 B:60.00 /60 B@:34 @:117
Recv: T:216.49 /220 B:60.00 /60 B@:34 @:153
Recv: T:217.02 /220 B:60.00 /60 B@:34 @:114
Recv: T:217.19 /220 B:60.00 /60 B@:34 @:151
Recv: T:217.72 /220 B:60.00 /60 B@:34 @:186
Recv: T:218.07 /220 B:60.00 /60 B@:13 @:185

wildcardfox, 2016-02-21:
Hmmm, Ryan have you tried Repetier server? I might try it today. I'm waiting for my usb hubs to arrive, but after reading your post it sounds like power is not the issue.

rcarlyle, 2016-02-22:
I don't really have time to try it at the moment, sorry. Just chipping in that I'm seeing the same thing.

wildcardfox, 2016-02-22:
No worries. Just chilling in gave me some insight and the fact that Hoya re using Octopi instead of Octoprint suggests that it is not something that is only occurring on Octoprint like I previously assumed. Just chiming in helps a lot. Thanks!

rcarlyle, 2016-02-22:
Well, it's Octoprint on a Raspberry Pi. Or is that what you meant?

So far, I've only seen it happen while doing M109 and M190 at the same time. M104 and M140 don't seem to do it. Could just be coincidence though.

wildcardfox, 2016-02-22:
yah sorry just realized that they are the same thing I got confused when reading on the Octoprint page where they specially say that they are different, but you are right Octopi is Octoprint for Raspberry Pi. 

BTW thank you Ryan for your post on Smoothieware and it's bug, I ran into that issue and found your the Delta bot rundown. 

I just installed the powered usb hub and I saw a small dip on the hotend while the heat bed was heating, but a constant curve for the hotend once it started to heat, which is awesome. That's a first in a long time. It's only one test/print, but perhaps the powered usb hub fixed the issue. I'll report back after a full day of printing. Hopefully Repetier was right and it was a just a power issue.

wildcardfox, 2016-02-22:
Okay. Update: I switched to a powered hub and switched over to Repetier Server, and the fluctuations when heating still persists.