repetier/RepetierFirmware #477: 0.92.8 megatronics 2.0 -- "Z-" input blink

phd0, 2016-02-20:
on a **megatronics 2.0**, the led on the end-stop flashes 0.1s every 1s. 
It do it with a optical end-stop and a mechanical 3 wires end-stop. 
And only if connected in the "Z-" input.
When i inverse external end-stop from 2 entry, the problem stay in the "Z-" input.
0.92.8 firmware, manual configuration.h from original. parameted with a full graphic smart controller (really connected or not)

repetier, 2016-02-21:
Does it also happen when you connect the x endstop to z endstop pin and z endstop to x endstop? This will test if the wiring matters or the endstop is simply defect.
What does M119 return? Does it also change with the led?
I'm no hardware guy so I do not know what makes a sensor led flash except what it is meant for. My guess is that the pin is defined as output, but that would mean a signal with 1hz frequency and we do not have this.

phd0, 2016-02-21:
1. on the MegaTronics 2.0
when I reverse the limit switches X and Y the problem remains on the input "Z-". 
I have tested with the industry "works092" is the same.
On the CONTROLLER_SMARTRAMPS screen, i see that the menu Position/Z position that the "Min endstop" change on/off without any physical action on the end-stop. no change on the "Max endstop" of the Z and in any other axes.  
with : 
`#define ENDSTOP_PULLUP_Z_MIN true`
`#define ENDSTOP_PULLUP_Z_MAX true`
`//set to true to invert the logic of the endstops`
`#define ENDSTOP_Z_MIN_INVERTING false`
`#define ENDSTOP_Z_MAX_INVERTING false`
`// Set the values true where you have a hardware endstop.`
`#define MIN_HARDWARE_ENDSTOP_Z true`
`#define MAX_HARDWARE_ENDSTOP_Z true`

2. on a RAMPS 1.4
I have tested all there end-stop on the same place +  on a RAMPS1.4 : 
the end-stop led don't blink, but it is exactly the same on the screen...

3. on a GeeeTech GT2560
I test on a GeeeTech GT2560 motherboard (with a only 2 wires end-stop) : work fine on the screen

phd0, 2016-02-26:
i work on it, but don't find anything...

closed 2018-04-24 by phd0