repetier/RepetierFirmware #487: Distortion Correction not correct

hannoHI, 2016-02-29:
Hi, I'm running a P3Steel (normal Cartesian printer) and noticed some problems when trying to use distortion correction.
I experience this with code synced today.
Output when manually probing the grid as define in the config results in strange behaviour.

extract from config.h:

#define DISTORTION_CORRECTION 1
#define DISTORTION_CORRECTION_POINTS 3
#define DISTORTION_CORRECTION_R 100
#define DISTORTION_PERMANENT 1
#define DISTORTION_UPDATE_FREQUENCY 15
//deliberately set START & END high to eliminate them from affecting the calculations
#define DISTORTION_START_DEGRADE 199
#define DISTORTION_END_HEIGHT 200

#define DISTORTION_EXTRAPOLATE_CORNERS 1
#define DISTORTION_XMIN 10
#define DISTORTION_YMIN 10
#define DISTORTION_XMAX 190
#define DISTORTION_YMAX 170

#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5

Autolevel=off
Results when probing:
With Correction matrix set to 0
(G33 R0)
16:08:46.612 : Info:Resetting Z correction
moved to point with G1 X10 Y10 Z25 then G30 to probe:
16:09:12.934 : Z-probe:23.18 X:10.00 Y:10.00 zCorr:0.00
16:09:18.622 : Z-probe:23.05 X:100.00 Y:10.00 zCorr:0.00
16:09:24.372 : Z-probe:23.28 X:190.00 Y:10.00 zCorr:0.00
16:09:31.497 : Z-probe:23.16 X:10.00 Y:90.00 zCorr:0.00
16:09:37.169 : Z-probe:23.02 X:100.00 Y:90.00 zCorr:0.00
16:09:42.872 : Z-probe:23.23 X:190.00 Y:90.00 zCorr:0.00
16:09:50.028 : Z-probe:23.26 X:10.00 Y:170.00 zCorr:0.00
16:09:55.778 : Z-probe:23.10 X:100.00 Y:170.00 zCorr:0.00
16:10:01.543 : Z-probe:23.27 X:190.00 Y:170.00 zCorr:0.00

Manually set each point of distortion correction to 0.1mm: (G33 -L0)
16:11:16.088 : G33 X10.00 Y10.00 Z0.100
16:11:16.088 : G33 X10.00 Y90.00 Z0.100
16:11:16.088 : G33 X10.00 Y170.00 Z0.100
16:11:16.088 : G33 X100.00 Y10.00 Z0.100
16:11:16.088 : G33 X100.00 Y90.00 Z0.100
16:11:16.103 : G33 X100.00 Y170.00 Z0.100
16:11:16.103 : G33 X190.00 Y10.00 Z0.100
16:11:16.103 : G33 X190.00 Y90.00 Z0.100
16:11:16.103 : G33 X190.00 Y170.00 Z0.100

Same points probed 0.3mm correction? (Bed coating set to 1.2mm)
16:11:33.916 : Z-probe:23.47 X:10.00 Y:10.00 zCorr:0.10
16:11:40.026 : Z-probe:23.34 X:100.00 Y:10.00 zCorr:0.10
16:11:46.166 : Z-probe:23.57 X:190.00 Y:10.00 zCorr:0.10
16:11:53.713 : Z-probe:23.47 X:10.00 Y:90.00 zCorr:0.10
16:11:59.807 : Z-probe:23.32 X:100.00 Y:90.00 zCorr:0.10
16:12:05.939 : Z-probe:23.53 X:190.00 Y:90.00 zCorr:0.10
16:12:13.501 : Z-probe:23.56 X:10.00 Y:170.00 zCorr:0.10
16:12:19.673 : Z-probe:23.40 X:100.00 Y:170.00 zCorr:0.10
16:12:25.861 : Z-probe:23.56 X:190.00 Y:170.00 zCorr:0.10

All I can think is that the steps/mm of my Z axis is too high? Is the correction stored in steps?
Z-steps/mm on my printer is 16000/mm

Any pointer to where in the code to look would be helpful.

PS. zprobe need very little force to work & is quite repeatable (+-0.01) when I did repeats of the above tests

repetier, 2016-02-29:
Well 16000 steps per mm is very high and quite unnecessary high. Anything above 1000 is more or less overkill.

Latest updated fixed a computation problem with distortion correction. That requires steps when correcting to be low enough to not create an overflow.
You start correction after 199 mm so that is already high (why 199, if you never want a plain z set it 3*maxZ to be on safe side that it never tries to interpolate.
If you want stop interpolating, make sure
correction_z = (correction_z * (zEnd - z)) / (zEnd - zStart);
dos not overflow. All values are in steps and z < zMax * 200*16000
Critical part is
correction_z * (zEnd - z) which must be < 2147483648
biggest z here is zStart
so zEnd*(zEnd-zStart) < 2147483648

So in your case 5*10^10 which is factor 20 to high.

So switch to 0 compensation near bottom which is the normal intension or switch never by setting start > 3 * zheight

BTW: You never said what your problems are. Just assumed it's a new problem and that is what I changed and what overflows in your case (maybe)

Hope then it works again.

hannoHI, 2016-02-29:
Thanks,
I will look into the overflow side of things.

Problem is I set z correction for distortion to 0.1 & the level seem to shift by 0.3mm.
I will have a look at the overflow.

So if I set the start to approx 1 & end to 2 and keep my print height below say 10mm it should work?

I will see how low I can drop the steps on my Z axis while I'm at it.

My zmax approx 180mm (so I set START & END > 180mm )
I think my distortion is due to mechanical problems & not my bed's levelling as such.

repetier, 2016-02-29:
latest updated fixed overflow problem. I also got a patch that should fix a problem with distortion but I need to understand it first.

hannoHI, 2016-02-29:
I used the latest patch, reduced START & END & replaced my stepper driver (unfortunately my RADDS 1.1 need hardware mod of my steppers to run lower step rate) but got the steps/mm now down to 8000. 
Even moving between probes now seem a lot smoother. I think something did overflow & caused some of the problems.
Testing still in progress but it seems that it now have an error of +-0.01 most of the time +-0.02 worst case - with 1 or 2 outlying points. - Much better than the +-0.12 prior to the correction.
Thanks. 
Will see if I can now get my probe & nozzle aligned & do a few test prints tomorrow.

closed 2016-02-29 by hannoHI