repetier/RepetierFirmware #495: Apply Bed Matrix

arzhurp, 2016-03-10:
Hello,
I tried to implement the bed leveling on my Kossel without success. After G32 S2, the probe do the job, but nothing is applied during the print. After G32 and S2 G33 i've got the same result.
I tried EEPROM resest, M323 S1 P1, G29, M322 S3, BED_LEVELING_METHOD 0/1...

Nothing happens...
Can you help me ?
Thanks

repetier, 2016-03-10:
Check what your log says. It writes if enabled and what the rotation matrix looks like und if distortion is enabled. You need software rotation here, not hardware rotation.

arzhurp, 2016-03-10:
Can you send me the G-Code list in order to make it work ? I've follow your website tutorial but  G29 is only doing a 3 points measurement, in your video it's making a grid !  
Maybe my config might help you ;-)
Thank you.
[RepetierConfiguration.txt](https://github.com/repetier/Repetier-Firmware/files/167432/RepetierConfiguration.txt)

repetier, 2016-03-11:
Sounds like you have not compiled with distortion correction enabled.

Also use G33 now instead of G29 as the replace,ent function works much better, see repetier.ino file for description.

After calibrating endstops you do
G32 S2
G33

and are done. What happens depends on your configuration settings.

arzhurp, 2016-03-13:
--> Sounds like you have not compiled with distortion correction enabled.
I have compiled with distortion correction enabled ;-)

--> Also use G33 now instead of G29 as the replace,ent function works much better, see repetier.ino file for description. 
You should remove G29 then ;-) 

I'll reset my eeprom, create a new config file, send it to the arduino and try from a fresh install to be sure ;-)