repetier/RepetierFirmware #496: Invert step pin logic

Megane008, 2016-03-10:
Hello,

Im using Toshiba external drivers for my steppers and these drivers have inverted logic on step pins. In Marlin there is an option in advaced config to invert this. I know there is no option in Repetier but I think its possible to change this manualy somewhere as HIGH/LOW. Could you please direct me where should I do it? Im using mixing Diamond so 3 extruder motors which are wired trough Toshiba drivers and there I need to invert this.

Thanks

repetier, 2016-03-10:
You can set STEP_WITH_HIGH accordingly. Works only globally so all drivers must be the same. Jusr add the define in manual tab of the config tool.

Megane008, 2016-03-10:
Thanks, and is it possible to do it not globally but only for Extruders?

repetier, 2016-03-11:
Not with a switch. Then you have to find allmoccurences in Extruder.h and cpp and invert it in the code.

Megane008, 2016-03-13:
Thanks, I found the occurences only in Extruder.cpp but not in Extruder.h
But I guess all I need to do is change !START_STEP_WITH_HIGH to START_STEP_WITH_HIGH and vise versa right?

repetier, 2016-05-12:
Exactly, just as your config requires it.

closed 2016-05-12 by repetier