repetier/RepetierFirmware #499: Strange jerky movements

Megane008, 2016-03-17:
Im having problem with Repetier, probably path planner. 
The printer moves without problems in dry run mode. But when I print it makes jerky movements. 
Increasing acceleration helps ( but it only helps because of higher velocity mass )
Im using delta printer with Ramps 1.4 board, standard onboard drivers for XYZ
using acceleration 1000, Jerk 20 ( triied modifing up and down but no help ) 
A bit improvement if I decrased extruder acceleration from 5000 to 1000 but still a bit jerky movement. 
Im using 3 extruders wired trough external stepper drivers Toshiba ( I had to manualy switch step HIGH/LOW in firmware to make them work )
I triied diferent temperature control even simple bang-bang but still the same. 

Any Idea what could be an issue?

repetier, 2016-03-17:
Hi,

delta printers with avr often have a speed problem as avr is qute slow. So frequency of steppers is limited to 14-20khz otherwise computation takes too much time. So if your extruder resolution is very high that gets a problem. In that case reduce extruder steps or use faster hardware like radds board with due which can handle 150khz.

Yours
Roland Littwin

Megane008, 2016-03-18:
Thanks Roland,
Any chance you could recomend me which electronics works best with Repetier? Cause I want to stick with your firmware and expand my 3 extruder diamond to 6 extruder hotend and so far no other firmware can do color mix as yours. Im looking for some fast electronics which can drive up to 6 extruders + 3 XYZ and Im able to use external drivers so I only need enough PWM pins.

jamesarm97, 2016-03-18:
I was going to recommend the RADDS board plus expansion but that would only get you 5 extruders. http://doku.radds.org/dokumentation/radds/

repetier, 2016-03-18:
You could still use radds board with extension board and one external motor driver. Extension board has 2 extra pins and adc oin on the side canalso be used, so you have enough pins for 6 motors and with diamond hotend you need no extra heaters so. that is ok.

Megane008, 2016-03-19:
Thanks for info I will definetly go for Radds but sadly they are out of stock everywhere in decent shops:/ will have to wait a bit.