repetier/RepetierFirmware #500: Auto Bed levelling Issue

pratiknarkhede123, 2016-03-18:
Hi there,
I am using printer with build platform size 500mm x 500mm and inductive proximity sensor. I am trying autobed levelling with Gcode G32 S2 with grid size 7 but i am getting error of .5 to 1 mm .
These are my printer settings:
https://onedrive.live.com/redir?resid=43641BEC443B251D!334&authkey=!AJ99HTuH1MIbmlA&v=3&ithint=photo%2cpng

Please provide me solution for this.

pratiknarkhede123, 2016-03-20:
Awaiting response from you.

repetier, 2016-03-20:
Where do you get which error? Is the bed plain. Especially big beds tend to be uneven and g32 does only make a average plane through the points. Add distortion correction with G33 to compensate that as well.

pratiknarkhede123, 2016-03-23:
when i am giving G33 command it shows unknown error. Is there anything else to be used with G33 like i am using S2 with G32.?