repetier/RepetierFirmware #501: repetier // M163, M164 codes are working or not??...

yhhwnsgur, 2016-03-19:
i send code "M163 S0 P1
                     M163 S1 P0
                     M163 S2 P0
                     M164 S0 "   in repetier host.

but this massage "unknown command"....

i follow the repetier color mixing page. i checked the codes in repetier firmware in arduino....

what shoud i do?

please  help me  ㅜㅜ

repetier, 2016-03-19:
These commands are only available if you compiled the firmware with mixing extruder support, so i guess you have not done this.

yhhwnsgur, 2016-03-21:
<repetier firmware>
<command.cpp>
#if MIXING_EXTRUDER > 0
    case 163: // M163 S<extruderNum> P<weight>  - Set weight for this mixing extruder drive
        if(com->hasS() && com->hasP() && com->S < NUM_EXTRUDER && com->S >= 0)
            Extruder::setMixingWeight(com->S, com->P);
        break;
    case 164: /// M164 S<virtNum> P<0 = dont store eeprom,1 = store to eeprom> - Store weights as virtual extruder S
        if(!com->hasS() || com->S < 0 || com->S >= VIRTUAL_EXTRUDER) break; // ignore illigal values
        for(uint8_t i = 0; i < NUM_EXTRUDER; i++)
        {
            extruder[i].virtualWeights[com->S] = extruder[i].mixingW;
        }

these codes means nothing?

so... i have to write some codes additionally??

repetier, 2016-03-21:
This code is only included with 
#define MIXING_EXTRUDER 1
and you seem to have set it to 0 in your configuration.h which means your printer ha sno mixing extruder.

closed 2016-05-12 by repetier