repetier/RepetierFirmware #503: Add support to filament width sensor

Nandox7, 2016-03-21:
Hi,

It would be great to have support in the Repetier firmware for a filament width sensor.
There are some new prototypes shown up that seem promising.

Example: http://forums.reprap.org/read.php?1,636034

The approach is usually the same, where a analog input is used.
Example from Marlin that has it implemented: https://github.com/MarlinFirmware/Marlin/blob/5a54204f723f7fd110b57d19f0de01cecd8d47e7/Marlin/temperature.cpp#L675

repetier, 2016-03-21:
I've put it on my long todos:-)

Nandox7, 2016-03-21:
Nice
That's a start. :)

I'm getting some parts to make a prototype and I'll into this to see if I can hack the code a bit to get some minimal functionally working. 
Worst case I just waste filament.