repetier/RepetierFirmware #505: capability to control the ready to use cutting modules such as auto focus laser head or plasma THC

mkeyno, 2016-03-22:
hi , I was wondering if this firmware have capability to control the ready to use cutting modules such as auto focus laser head or plasma THC , 
I saw couple of effort on Marlin enthusiast to check output sensor the cutting tool and control the stepper motor the module to control height during the printing 
any comments really appreciated

repetier, 2016-03-22:
I do not have such hardware and even do not know what exactly you are talking about. All I have is a simple china laser I can switch on/off.

The laser code is written to be exchangeable so it is possible to change it to use such hardware if you know how to control it.

mkeyno, 2016-03-22:
thanks @repetier  for reply , but unlike the Marlin , I couldn't find the modified RepRap for CNC cutting machine such as laser or plasma , and it seems unlike the 3d Printer  most developer here have own way and not follow the standard same way for CNC cutting machine
actually I made 120 watt  laser CNC with RAMPS and modified Marlin by [ TurnkeyTyranny]( https://github.com/TurnkeyTyranny ) which is equipped with BT wireless link , I used to  repetier host for cutting and engraving. Code has minor defect  and unfortunately not updated any longer , however I'm moving to next step and find many industrial laser CNC have  module which is keep the cutting tool; in readiness state , for example THC module check the correct distance from material constantly , I start sending the note to Marlin and repetier developer to find any enthusiast people to such project and make standard modified version for cutting  CNC machine
I really appreciated to have your comments

mkeyno, 2016-03-28:
dear @repetier  can I have your idea about possibility of auto height control of CNC tools in the RRF? 
Right now Manually  Control the laser head height is possible with Marlin function such as [action_laser_focus_custom(),](https://github.com/HakanBastedt/Marlin4Due/blob/Laser/Marlin/ultralcd.cpp#L106-L126)  and I think ,in two ways we can control the  tools heights  during printing our cutting
One with help  of separate modules and one is totally implemented  in firmware
In the first method (hard way) we could use the Arduino module to command the z axis whenever new bias with pre set  g-code height come out, and when adjustment performed active the readiness pin  which is controlled by g-code M266 command before each  cutting   
In the second way we could use the function such as [laser_set_focus(currentZ)](https://github.com/HakanBastedt/Marlin4Due/blob/Laser/Marlin/ultralcd.cpp#L1208) in loop function , in this case we need use I2C to request the height from sensor module    

also I really appreciated to  have your colleagues comments