repetier/RepetierFirmware #506: Watchdog depends on SD card status (AVR)

osnwt, 2016-03-22:
When AVR version with watchdog enabled tries to mount SD card (particularly, after power up), and SD card is missing, it takes few seconds. Since watchdog starts before SD card mount, this constantly reboots the board. This may or may not affect the SD card operation after boot (it will not if watchdog is reset in an interrupt).

**Workaround**: insert SD card to boot and/or flash new firmware.
**Solution**: fix the code to enable watchdog reset during SD card mount (on boot, at least) or reorder the init sequence to enable watchdog only after SD card probe (assuming it will be reset later in an interrupt).

jamesarm97, 2016-03-23:
Watchdog I it was moved once after SD in it for this reason. I guess Init and actual load might be two different things. See commit https://github.com/repetier/Repetier-Firmware/commit/336ac86d33beb2aafc54fa43368c4f0505a9a1b0

osnwt, 2016-03-23:
Agreed, but still if watchdog is enabled and no SD card inserted, the system can't boot (restarts).
Also after the system booted, if you remove the SD and try to mount it again, the firmware also restarts. Probably, SD code needs one more watchdog kick somewhere :-)

rcarlyle, 2016-03-23:
Have to be careful with watchdog reset calls in the SD code -- the SD card library can lock up the firmware and cause a thermal runaway. You really want zero watchdog reset calls outside the heater management code if at all possible.

osnwt, 2016-03-23:
I've looked at the commit mentioned above. That was related to SAM MCU only, and in that place the heaters are not enabled yet. I agreed also about watchdog resets (there are many scattered in the code).

In any case it is a bug that has to be either fixed, or documented as a known issue. Currently web configurator suggests to enable watchdog. But if enabled, some AVR users can be in trouble. Some AVR boards are shipped with bootloaders which do not disable watchdog at all (my Melzi is an example, and there were complains on forum recently). Having any issue with SD, users can't update boards because can't enter the bootloader mode (it will be reset again and again by watchdog). The only solution (apart from inserting the SD - and they have no idea about the reason of reboots) is to power off/on the board and enter boot mode in a second. Not an easy trick if you do not know what are you doing.

repetier, 2016-03-24:
@rcarlyle Thermal runaway is not possible any more. To trigger a ping you need the normal HAL::ping + thermal control loop must be run. With this and condition we assure that temp. settings are updated.

I checked the enable watchdog for avr and found out that Mega2560 and 1280 AVR set watchdog to 4s and all others, especially melzi/teensy to 1s. I have set that now also to 4s in master and online config as I never had problems with sd card.

Please report if this still causes reboot without sd card inserted. With the extra 3s that should not happen any more.

rcarlyle, 2016-03-24:
@repetier I don't want to hijack this issue discussion, but could you link me to the code or explain a little more what you mean? Thanks

jamesarm97, 2016-03-24:
My guess is he sets a flag in foreground calls to say to refresh the watchdog since it is still alive and running but doesn't send the watchdog write in the interrupt (timer) routine unless that flag is set. That way the interrupt routine will not ping the watchdog if the main code is not running right.

This is at the very end of the interrupt (timer) / heater controlling routine before the interrupt returns.

#if FEATURE_WATCHDOG
  if(HAL::wdPinged) {
     WDT->WDT_CR = 0xA5000001;
     HAL::wdPinged = false;
  }

repetier, 2016-03-24:
Exactly HAL::pingWatchdog(); only sets HAL::wdPinged = true while the interrupt executes the real ping. That makes sure interrupts get still called. During normal print only manageTemperatures send a ping so we know both got called lately.

rcarlyle, 2016-03-24:
Got it, thanks.

bartanderson, 2016-11-27:
I didn't see watchdog to enable on any of the pages for my configuration. Does that mean we can get rid of the warning on the download page or conditionally remove it based on configuration?

The text I am referencing is "Please notice the warnings below, which normally indicate a wrong or bad choice of configuration combination.

No watchdog activated. In case of a firmware hang this can cause damage and fire to your printer! And even a 100% correct firmware can hang if voltage drops or electrical noise is near."

repetier, 2016-11-28:
No, in config tool it is in features tab 5th checkbox from top. So it still makes sense.