repetier/RepetierFirmware #516: Strange behaiviour on external steppers on Extruders

Megane008, 2016-04-15:
Hello,

I just upgraded to Due with Radds 1.5. I got last 3 Raps 128 they had on store with it for XYZ but for extruders I still have to use my old external Toshiba 6128 drivers. It worked nicely on Ramps 1.4 but on Radds 1.5 I cant get it running properly. I even tried A4988 and same issue. The problem is that signal is going way too fast for stepper drivers to manage so motors dont move. When I set High delay on 50 ( yes its way too much ) I get them moving but strange thing they move same speed no matter if I change steps per mm or if I do G1 E100 F1000 or G1 E100 F10 still same. Any idea what could be wrong? 
Thanks for help

repetier, 2016-04-15:
Stepper high delay 1 should suffice for A4988, not sure how many us 6128 take. With delay 50 you get a maximum for 20000hz if we neglect all other time consuming tasks, so 16000 is more realistic. So speeds are just limited by the delay which is not taken into account of your computations.

Did you set steps per mm etc in eeprom?

What speed do they move and what are your steps per mm?

Megane008, 2016-04-15:
Well I set steps per mm in eeprom and when it didnt do any physical change I changed them also in compilation and reuploaded. I go for 107 steps per mm. But seting 1 step per mm shows no change as well as well as setting 1000 steps per mm. Which is rly strange because it should make diference. Im also using repetier host and doing manual extrusion but I guess that cant be it. 

The speeds correspond to 100 mm/s but in repetier host is set manual extrusion speed on 2 so I dont get it. 

But:-) when I plug in Raps 128 it moves absolutely normaly so the issue is with drivers.