repetier/RepetierFirmware #519: comment of uint8_t PrintLine::queueNonlinearMove()

voicevon, 2016-04-19:
Location: motion.cpp
uint8_t PrintLine::queueNonlinearMove()

Comment of the third param 
@param delta_step_rate delta step rate in segments per second for the move.
It doesn't match "uint8_t softEndstop"

repetier, 2016-04-19:
Sorry, but what do you want to say here? third param is softEndstop but what is the problem?

voicevon, 2016-04-20:
Sorry, I didn't clearly express myself
The following  is the description of the third param in comment 
        @param _**delta_step_rate delta step rate in segments per second for the move.**_
Now , let's take a look of function definition:
uint8_t PrintLine::queueNonlinearMove(uint8_t check_endstops,uint8_t pathOptimize, **_uint8_t softEndstop_**)

They are  not the same name. I guess the comment description is wrong.
Thanks.

repetier, 2016-04-20:
Ok, now I understand. Fixed for next release.

closed 2016-04-20 by repetier