repetier/RepetierFirmware #520: Slowing down XY axes does not work!

martincote418, 2016-04-23:
Hello,

I want to groove PCB with my Wanhao I3 Duplicator at very low speed.

I try to slowing down (very slow IE F12) XY axes without success. It work for the Z axis but not for XY. I know I can set EEPROM Feed rate values and it works but speed also applies to the positioning of the head which makes the work much longer. I would like the positioning is done at a normal speed, but the movement during cutting is done very slowly.

I can update the Repetier firmware but I do not know which value to slowing down the XY speed.

Someone can help me please? I am new in this world :)

Martin

repetier, 2016-04-23:
Due to explicit integration of speeds during acceleration we enforce a minimum speed to avoid numerical problems. This depends on acceleration for x and y. So reduce xy acceleration in eeprom and you can reach lower speeds.

martincote418, 2016-04-29:
I can't believe it is so simple. I spent lot of time to achieve that! I am newbie in this world. I had to reduce travel acceleration too to reach very slow speed. Thank a lot for you quick answer. :)