repetier/RepetierFirmware #521: Z min endstop ignored / random data

sigxcpu76, 2016-04-27:
Hi,

I have an issue that I simply cannot address. The Z_MIN endstop is ignored or displays random data. It looks like it is reading another pin, not the preset one (motherboard 33 RAMPS, pin 18).
I've put Marlin, it works. I've made a small sketch that displays D18 status and it works. The issue occurs with Repetier FW only.

I've attached the Configuration.h because I'm pretty sure I'm missing some setting.

```c
/*
    This file is part of Repetier-Firmware.

    Repetier-Firmware is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Repetier-Firmware is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Repetier-Firmware.  If not, see <http://www.gnu.org/licenses/>.

*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/* Some words on units:

From 0.80 onwards the units used are unified for easier configuration, watch out when transferring from older configs!

Speed is in mm/s
Acceleration in mm/s^2
Temperature is in degrees celsius

##########################################################################################
##                                        IMPORTANT                                     ##
##########################################################################################

For easy configuration, the default settings enable parameter storage in EEPROM.
This means, after the first upload many variables can only be changed using the special
M commands as described in the documentation. Changing these values in the configuration.h
file has no effect. Parameters overriden by EEPROM settings are calibration values, extruder
values except thermistor tables and some other parameter likely to change during usage
like advance steps or ops mode.
To override EEPROM settings with config settings, set EEPROM_MODE 0

*/

// BASIC SETTINGS: select your board type, thermistor type, axis scaling, and endstop configuration

/** Number of extruders. Maximum 6 extruders. */
#define NUM_EXTRUDER 1

/** Set to 1 if all extruder motors go to 1 nozzle that mixes your colors. */
#define MIXING_EXTRUDER 0

//// The following define selects which electronics board you have. Please choose the one that matches your setup
// Gen3 PLUS for RepRap Motherboard V1.2 = 21
// MEGA/RAMPS up to 1.2       = 3
// RAMPS 1.3/RAMPS 1.4        = 33
// Azteeg X3                  = 34
// Azteeg X3 Pro              = 35
// Ultimaker Shield 1.5.7     = 37
// Gen6                       = 5
// Gen6 deluxe                = 51
// Sanguinololu up to 1.1     = 6
// Sanguinololu 1.2 and above = 62
[... 1678 lines cut ...]
#define USER_KEY1_PIN     UI_DISPLAY_D5_PIN      // D5 to display (not used for graphics controller), change to other pin if you use character LCD !
#define USER_KEY1_ACTION  UI_ACTION_FAN_SUSPEND
#define USER_KEY2_PIN     UI_DISPLAY_D6_PIN      // D6 to display (not used for graphics controller)...
#define USER_KEY2_ACTION  UI_ACTION_SD_PRI_PAU_CONT
#define USER_KEY3_PIN     UI_DISPLAY_D7_PIN      // D7 to display (not used for graphics controller)...
#define USER_KEY3_ACTION  UI_ACTION_LIGHTS_ONOFF
#define USER_KEY4_PIN     -1
#define USER_KEY4_ACTION  UI_ACTION_DUMMY
*/

// ####### Advanced stuff for very special function #########

#define NUM_MOTOR_DRIVERS 0
// #define MOTOR_DRIVER_x StepperDriver<int stepPin, int dirPin, int enablePin,bool invertDir, bool invertEnable>(float stepsPerMM,float speed)
#define MOTOR_DRIVER_1(var) StepperDriver<E1_STEP_PIN, E1_DIR_PIN, E1_ENABLE_PIN, false, false> var(100.0f,5.0f)

/*
  You can expand firmware functionality with events and you own event handler.
  Read Events.h for more informations. To activate, uncomment the following define.
*/
//#define CUSTOM_EVENTS

#endif

```

MarPiRK, 2016-04-27:
I had the same problem yesterday. It seems like it's somehow wired to serial port. I've worked around this by changing Z_MIN pin in `pins.h` to 2 (which is X_MAX on RAMPS). It's just workaround and it definitely needs some investigation from some dev.

jamesarm97, 2016-04-27:
It looks like this line “could” be the problem. I don’t think you want to define a bluetooth com port of 1 since it would interfere with the D18 pin. Set it to –1 unless you are really using a bluetooth module, then you would need to figure out what other pins you could use instead.

#define BLUETOOTH_SERIAL   1  (set to –1)                    // Port number (1..3) - For RUMBA use 3
#define BLUETOOTH_BAUD     115200                 // communication speed

sigxcpu76, 2016-04-27:
I've suspected it is something related to the Z_MIN pin, because remapping it to 19 (Z_MAX) worked.

repetier, 2016-04-27:
Is this one of these cheap RAMPS bundles? Heard that one of them has problems with crosstalk for some pin, so maybe it was the z min pin. So it all depends on timings and retesting but using a different pin is surely the best solution if result switches by external signals.

sigxcpu76, 2016-04-27:
Yes, it is an MKS BASE 1.4. Is it cross-talk? Because setting BLUETOOTH_SERIAL to -1 fixed it.
Also, as I've stated, running Marlin or a simple sketch that displays D18 changes (input mode, pull-up enabled) works reliably.
I see that D18 is the same pin as TXD1, so doesn't look like cross-talk to me, but pin sharing.
I think that Repetier should have a SanityCheck like Marlin has (for example BLUETOOTH_PORT == 1 and Z_MIN_PIN == 18 should result in a compile error). Poor man's test units :)

repetier, 2016-04-27:
If you had bluetooth on that pin it is no crosstalk. Then it is the copy of the output data to bluetooth and a result of configuration error.

sigxcpu76, 2016-05-04:
@repetier I will also close this issue, if that's ok. It was a user/configuration error that got solved.

repetier, 2016-05-04:
Sure.

closed 2016-05-04 by repetier