repetier/RepetierFirmware #523: 1/2 jerk and initial speed

sigxcpu76, 2016-04-27:
This is my issue, not Repetier's, I think.

My printer has issues with the initial "safe" speed. Marlin works fine because, as far as I know, has no "real jerk". But Repetier has and the initial/safe speed is 1/2 of Vxy Jerk.
On many sudden moves this makes the printer skip steps and even on normal prints that work there are strange "shock" noises heard during print. 
I've changed the "safe" speed to 0.1 * jerk instead of 0.5 *, I will test and see if this solves the skipped steps issue and the strange noises coming from the carriage.

Would using 10% of jerk be an issue except longer printing times?

repetier, 2016-04-27:
If the initial speed is too high you must reduce jerk. Marlin does the same as far as I know.

Jerk = possible speed change between direction changes. So with initial speed = +Jerk/2 it is allowed to reverse direction without acceleration to -Jerk/2. If starting with jerk/2 looses steps the reversing of direction will surely do the same, so only real solution is to reduce jerk completely.

In new versions you might try
#define REDUCE_ON_SMALL_SEGMENTS
which will slowdown if you have sever direction changes in a short distance which is nearly the same as adding several jerks in a short sequence.

There is also a option
#define ALTERNATIVE_JERK
for a more logical jerk computation.

Both helped for my printer. Starting from nothing is normally the easier part. Direction changes is where you normally lose steps.

jamesarm97, 2016-04-27:
Thanks for this info. I printers sound and move horribly when they do small circles (broken down into small segments) and sounds like it could be related to the last two.

If the initial speed is too high you must reduce jerk. Marlin does the same as far as I know.

Jerk = possible speed change between direction changes. So with initial speed = +Jerk/2 it is allowed to reverse direction without acceleration to -Jerk/2. If starting with jerk/2 looses steps the reversing of direction will surely do the same, so only real solution is to reduce jerk completely.

In new versions you might try
#define REDUCE_ON_SMALL_SEGMENTS
which will slowdown if you have sever direction changes in a short distance which is nearly the same as adding several jerks in a short sequence.

There is also a option
#define ALTERNATIVE_JERK
for a more logical jerk computation.

Both helped for my printer. Starting from nothing is normally the easier part. Direction changes is where you normally lose steps.

sigxcpu76, 2016-04-27:
Thank you for this information. I've set REDUCE_ON_SMALL_SEGMENTS in Configuration.h, but there are still jerk noises coming from the bearings (I think). I'm using Vxy jerk = 5. In Marlin with this set much higher, to 20, these noises never occur. The acceleration settings are the same, same mechanics. I'm pretty sure that is something Repetier specific (the segments from circles sound different too), but I don't understand what is the source to explain better.
My guess is that the move starts that sudden that the carriage tends to rotate a little (the belts are not tied exactly in the middle between the two rods). This twist makes the cheap LM8UUs unhappy. I've also used some cheap Igus bushings that worked better, but those had backlash regardless of installed firmware.
Is there any (other) setting that makes the acceleration/initial speed behave more like Marlin?

repetier, 2016-04-28:
Since you come from marlin, did you change things in eeprom or in configuration.h?
You change such things in eeprom and these values might still be too high.

Computations are very similar to marlin, so that should not be the problem.

sigxcpu76, 2016-04-28:
Actually I don't come from Marlin, but switch between them few times a day :)
Repetier has a better usage of the GCLD display and smoother operation with the same display. Marlin stutters from time to time since upgrading from 2004 text display. It also integrates better with Repetier Server which I like (e.g. EEPROM settings).
The steps are the same, of course, the max accelerations are the same (1500 for X, 1000 for Y), Jerk is 20 in Marlin, 5 in Repetier because of the reasons stated above.
The EEPROM is cleared with the EEPROM erase sketch between switching the firmwares, to make sure no junk is there.
Basically, the only real issue I have now with Repetier is that the movement starts with loud bang sometimes. On some models, when this happens a lot in a short time, there is a risk of missing steps, as happened before. Increasing motors current to reduce this risk is not an option as they become hot after few minutes and either them will lose torque or the driver may decide it needs cooling, so I'll reach the same steps missed outcome.

I don't know if it is related to how Repetier handles the motion, but it is 100% reproductible. There are some extra soft clicks coming from the motors when doing arcs/coordonated XY motion lines which doesn't happen in Marlin. Maybe this is related to "To reduce this jittering, the Repetier-Firmware takes half steps in the primary direction, if the speed is not too high (until 8000 Hz steprate for 16MHz CPU). That way, the driven axis are updated earlier, reducing the jittering. In the example above, the green dots show earlier updates." from here: https://github.com/repetier/Repetier-Firmware/wiki/Hardware-settings-and-print-quality

repetier, 2016-04-28:
The half step handling was removed in recent versions. So questions should first be what version are you using? I always refer to latest version (developement which is currently nearly exact 0.92.9)

Next question, what kind of printer? Cartesian/core/delta?

Since it is repeatable, can you write a very small gcode that causes the sound to appear. If you have one with 2-5 lines and compile firmware with
#define #define DEBUG_QUEUE_MOVE
you can run it with echo on. Log then shows exact data on which speeds and acceleratios are used. That shoudl solve the mystery.

sigxcpu76, 2016-04-28:
I use the master branch since 3 days ago, so "nearly 0.92.9".
The printer is Cartesian, P3 Steel frame, (cheap)LM8UU bearings on axes.

I did a few tests in the past with very simple code like move 10mm back and forth very quickly, move 2mm back and forth for faster movements and it always worked without a single bang. So, the main issue first is to attempt to identify what are the exact move patterns that results in the bang.

I didn't try to do small quick squares and circles. Will do that when I'll switch back to Repetier.

repetier, 2016-04-28:
Great. As soon as you have a gcode causing the bang I'll try myself. If I also get a bang I might find it myself, otherwise the echo solution should hopefully give some insight.

sigxcpu76, 2016-04-29:
```c
// Disables axis when it's not being used.
#define DISABLE_X 0
#define DISABLE_Y 0
#define DISABLE_Z 0
#define DISABLE_E 0
```

This is unclear. Are these supposed to be disabled after some time unused or immediately after the move? I was having this enabled for X, Y and E before. Now I've set them to 0 and I've failed to hear the strange sounds yet.

repetier, 2016-04-29:
They get disabled directly. For timeout there is a general disable motors after timeout setting. Normally you only use z if that does not move when off. Fox xy it might be a problem as you see. Never thought about that setting.

sigxcpu76, 2016-04-29:
Actually Z was the only one with disable == 0 on my printer, because I'm using lead screws with large pitch (8mm dia, 8mm pitch) so I don't trust them to not rotate on vibrations.
I'll do some prints these days and come back with feedback.

sigxcpu76, 2016-05-03:
Back here to confirm that setting DISABLE_* for axes was the cause for both "clicking" sounds during arcs and sudden motions that lead to missed steps.
Maybe a suggestion, mention this in the vanilla Configuration.h for others to not hit the same issues as me.

Thanks everybody, especially @repetier for the great firmware.

closed 2016-05-03 by sigxcpu76