repetier/RepetierFirmware #528: hi,DUAL Z  ENDSTOPS

linjinc, 2016-05-21:
Z_DUAL_ENDSTOPS is a feature to enable the use of 2 endstops for both Z steppers - Let's call them Z stepper and Z2 stepper.
  That way the machine is capable to align the bed during home, since both Z steppers are homed.
 
  After Z homing, this adjustment is applied to just one of the steppers in order to align the bed.
  One just need to home the Z axis and measure the distance difference between both Z axis and apply the math: Z adjust = Z - Z2.
  If the Z stepper axis is closer to the bed, the measure Z > Z2 (yes, it is.. think about it) and the Z adjust would be positive.
  Play a little bit with small adjustments (0.5mm) and check the behaviour.

repetier, 2016-05-21:
Nice explanation, but what is the issue? Have you implemented this feature? Would be great as I have no printer with such a setup.

linjinc, 2016-05-21:
https://github.com/MarlinFirmware/Marlin
marlin is ok

I have a machine.
dual z endstop
[image]

repetier, 2016-05-21:
Ok, but what is your intention of the post?

linjinc, 2016-05-21:
I want to use repetier firmware
But repetier firmware does not have  dual Z endstop

repetier, 2016-05-22:
Ok, so you want me to implement this feature. In theory it is simple but having no printer for testing I never implemented it. If you are up to testing it until it works as expected I can write the modifications. Might be some tests required to get it going right. So say only yes if you will do it to the final end.

linjinc, 2016-05-22:
thank you
ok,

repetier, 2016-05-22:
Ok, here we go. Please use the latest development version for this. In config tool add in manual tab the following

#define MINMAX_HARDWARE_ENDSTOP_Z2 1
#define Z2_MINMAX_PIN 16
#define ENDSTOP_Z2_MINMAX_INVERTING 0
#define ENDSTOP_PULLUP_Z2_MINMAX 0

of course adjusted for your endstop type and pins. That should do the trick. M119 should show 2 z endstop values now. Test both and test homing. Hold a hand on power on first test if it does not stop, but I'm quite confident that I captured the right logic.

linjinc, 2016-05-22:
thank
I am here is the early morning, tomorrow test

linjinc, 2016-05-23:
Normal working
But I'll be testing a little more.

linjinc, 2016-05-23:
Z1 Morot and Z2 motor is not parallel
Z1 Sometimes pause
Z2 no problem

linjinc, 2016-05-23:
[image]
[image]
[image]

repetier, 2016-05-23:
What do you mean with z1 sometimes pauses? During a single move it stops and goes? That could be from friction if they are not parallel. The only big question is if they are not parallel, if only one motor stops and the other moves a bit longer. If you tried that and have to much difference the friction can make motors stall.

You could try with slower homing speed for testing to see difference.

linjinc, 2016-05-23:
Please forgive me for using Google Translation
May be set incorrectly, I will continue to test

linjinc, 2016-05-24:
I'm starting to use it, everything is okay.
Thank

closed 2016-05-24 by linjinc

Thump2010, 2016-06-07:
I had not used repetier before due to the lack of dual z endstop support. I did what is listed here, I have both motors moving in the right dir, I have both endstops able to trigger, I can see them trigger with M119 but when I home the Z axis it ignores the trigger and just keeps going so I have to power off.

shaiss, 2016-09-29:
@linjinc can you share a bit more on how you set this up?  What board and where is your 2nd z motor plugged into?

siggi60, 2017-07-12:
Hi and sorry for gravedigging,
ist it also easy to setup a third Z-Endstop? I don't really believe that all is done by adding some lines like them:

#define MINMAX_HARDWARE_ENDSTOP_Z3 1
#define Z3_MINMAX_PIN 16
#define ENDSTOP_Z3_MINMAX_INVERTING 0
#define ENDSTOP_PULLUP_Z3_MINMAX 0

My intention is it might be better to use three optical or hall endstops then one inductive sensor to do mabl.

Thanks for reading