repetier/RepetierFirmware #533: Z-Probing and X, Y Endstop bug 92.9

SanderVocke, 2016-05-30:
I think I saw someone mention this same bug somewhere, but I was unable to find it back.

On the G32 command, a cartesian printer first homes x and y, then calls startProbing(). startProbing performs another move to compensate for the Z probe position offset (moveRelativeDistanceInSteps).

This is a problem because it may cause it to move to negative positions in X and Y, depending on the offset. It happened to me: printer slams into the X endstop because it is trying to move from X=0 to X=-offset.

I fixed it for myself by pre-moving the printer to a safe spot inside Printer::startProbing(). Not sure how this would be done neatly.

SanderVocke, 2016-05-30:
My quick fix in BedLeveling.cpp: 

`#if FEATURE_Z_PROBE
void Printer::startProbing(bool runScript) {
    float oldOffX = Printer::offsetX;
    float oldOffY = Printer::offsetY;
    float oldOffZ = Printer::offsetZ;
    if(runScript)
        GCode::executeFString(Com::tZProbeStartScript);
    float maxStartHeight = EEPROM::zProbeBedDistance() + (EEPROM::zProbeHeight() > 0 ? EEPROM::zProbeHeight() : 0) + 0.1;
    if(currentPosition[Z_AXIS] > maxStartHeight) {
        moveTo(IGNORE_COORDINATE, IGNORE_COORDINATE, maxStartHeight, IGNORE_COORDINATE, homingFeedrate[Z_AXIS]);
    }
    Printer::offsetX = -EEPROM::zProbeXOffset();
    Printer::offsetY = -EEPROM::zProbeYOffset();
    Printer::offsetZ = 0; // we correct this with probe height
	
	//check that we will not end up at negative positions
	float toMoveX = Printer::offsetX - oldOffX;
	float toMoveY = Printer::offsetY - oldOffY;
	
	if((currentPosition[X_AXIS] + toMoveX) < 0.0f){
		float preMovePos =  1.0f - Printer::offsetX - toMoveX;
		moveTo(preMovePos, IGNORE_COORDINATE, IGNORE_COORDINATE, IGNORE_COORDINATE, EEPROM::zProbeXYSpeed());
		Commands::waitUntilEndOfAllMoves();
	}
	
	if((currentPosition[Y_AXIS] + toMoveY) < 0.0f){
		float preMovePos =  1.0f - Printer::offsetY - toMoveY;
		moveTo(IGNORE_COORDINATE, preMovePos, IGNORE_COORDINATE, IGNORE_COORDINATE, EEPROM::zProbeXYSpeed());
		Commands::waitUntilEndOfAllMoves();
	}
	
    PrintLine::moveRelativeDistanceInSteps(toMoveX * Printer::axisStepsPerMM[X_AXIS],
                                           toMoveY * Printer::axisStepsPerMM[Y_AXIS],
                                          0, 0, EEPROM::zProbeXYSpeed(), true, ALWAYS_CHECK_ENDSTOPS);
}

void Printer::finishProbing() {
    float oldOffX = Printer::offsetX;
    float oldOffY = Printer::offsetY;
    float oldOffZ = Printer::offsetZ;
    GCode::executeFString(Com::tZProbeEndScript);
    if(Extruder::current) {
        Printer::offsetX = -Extruder::current->xOffset * Printer::invAxisStepsPerMM[X_AXIS];
        Printer::offsetY = -Extruder::current->yOffset * Printer::invAxisStepsPerMM[Y_AXIS];
        Printer::offsetZ = -Extruder::current->zOffset * Printer::invAxisStepsPerMM[Z_AXIS];
    }
	
	//check that we will not end up at negative positions
	float toMoveX = Printer::offsetX - oldOffX;
	float toMoveY = Printer::offsetY - oldOffY;
	float toMoveZ = Printer::offsetZ - oldOffZ;
	if((currentPosition[X_AXIS] + toMoveX) < 0.0f){
		float preMovePos = 1.0 - Printer::offsetX - toMoveX;
		moveTo(preMovePos, IGNORE_COORDINATE, IGNORE_COORDINATE, IGNORE_COORDINATE, EEPROM::zProbeXYSpeed());
		Commands::waitUntilEndOfAllMoves();
	}
	if((currentPosition[Y_AXIS] + toMoveY) < 0.0f){
		float preMovePos = 1.0 - Printer::offsetY - toMoveY;
		moveTo(IGNORE_COORDINATE, preMovePos, IGNORE_COORDINATE, IGNORE_COORDINATE, EEPROM::zProbeXYSpeed());
		Commands::waitUntilEndOfAllMoves();
	}
	
    PrintLine::moveRelativeDistanceInSteps(toMoveX * Printer::axisStepsPerMM[X_AXIS],
                                           toMoveY * Printer::axisStepsPerMM[Y_AXIS],
                                           toMoveZ * Printer::axisStepsPerMM[Z_AXIS], 0, EEPROM::zProbeXYSpeed(), true, ALWAYS_CHECK_ENDSTOPS);
}
`

repetier, 2016-05-31:
The problem is somewhat solved in 1.0dev as it will not move into boundaries. But if the resulting position is illegal you still have a wrong measurement so it is important to select probing points that are possible to probe including offset.

SanderVocke, 2016-05-31:
Nice to hear it was fixed in the dev version!

I just want to clarify, as I'm not sure we're talking about the same problem.
In my case, all probing points were fine including the offset. The problem occurs before the probe even moves to the first probing point, because the sequence is "home -> compensate offset -> move to first point" - so the position can become negative in the "compensate offset" step.

Also, I am wondering what the ALWAYS_CHECK_ENDSTOPS does in the relative move:

PrintLine::moveRelativeDistanceInSteps(toMoveX * Printer::axisStepsPerMM[X_AXIS],
                                       toMoveY * Printer::axisStepsPerMM[Y_AXIS],
                                       toMoveZ * Printer::axisStepsPerMM[Z_AXIS], 0, EEPROM::zProbeXYSpeed(), true, ALWAYS_CHECK_ENDSTOPS);

It sounds like the move should check endstops to make sure it doesn't go out of bounds, but this doesn't (or didn't, not sure about the dev version) happen.

Thanks for the feedback!

repetier, 2016-06-01:
Yes, problem was in startProbing function where we now added boundary checks, so if it still does not work the fix should be there adjusting x/yExtra move.

The last parameter ALWAYS_CHECK_ENDSTOPS is from config and could be true or false. If true it will test hardware endstops for hit and stop move.