repetier/RepetierFirmware #534: RADDS LCD not working with latest development release

jamesarm97, 2016-05-31:
Was any work done on the UI / LCD since 0.92.8? I recompiled with my same configuration imported into the online configuration utility with feature_controller set to 7 and now the LCD does not work. Even resetting the board does not change anything on the screen. When I get a chance I will start checking out each version and seeing where it breaks at. The knob and sounds appear to work.

repetier, 2016-05-31:
My latest information is that the code is perfect and only the level shifter may cause troubles on some displays. If you can, try using the expansion port for the display signals and it should be very stable.

repetier, 2016-06-01:
Just found the mail for the problem with a alternative solution:

Ok, I found the cause to my problem. The level shifter on RADDS (TXB0106) tends to oscillate pretty easily. Adding 1KOhm resistors to the 3 signals LCD_{E,RS,4} after the level shifter solved it.

 

Others using the same configuration (RADDS v1.5 + RepRapDiscount Full Graphics Display) may have encountered something similar. After I solved it for myself, I found a thread in the RepRap forum where they suggest resistors as well, but they argue it's due to wrong 0 level which could as well be the same oscillation effect if they only measured it with a multimeter. Without the resistors I saw a perfect 50-60MHz sine wave on my oscilloscope and the +5V rail had a 50MHz 2Vpp ripple, i.e. 4-6V.

jamesarm97, 2016-06-01:
I will give that a shot. Just curious why the LCD works on 0.92.8 but on 1.0.0 it isn't. I previously was using a longer cable and it wasn't working but tried a trick that worked for me that involved moving the RS and E lines to the other end of the cable away from power and ground or something like that and the longer length started working but nothing I do on the latest dev version works. I even tried bumping up the longer delays that the RADDS define seems to add in the lcd write code.

repetier, 2016-06-01:
We haven't changed LCD handling for quite a while, I'd say at least since 0.92.8 so this also wonders me.  But that error was always hard to find. Till now all my displays work without tricks and others having same display have problems. So I guess the oscillations differs from board to board etc.

jamesarm97, 2016-06-03:
Stupid mistake. Remember me mentioning I moved a copy of the wires around on the ribbon cable to get rid of noise? I had to modify the ui.h file for the moved pin numbers and forgot to make those changes in the latest copy. Question, if I add UI_DISPLAY_RS_PIN 46 to the configuration.h file will it override the ui.h pin definitions? Should I undef then define in configuration.h?

closed 2016-06-03 by jamesarm97

repetier, 2016-06-03:
No you can not set it in configuration.h if you select a display type. Then everything gets overwritten from the display/board combo.