repetier/RepetierFirmware #535: Z-Probing 2mm offset after Z-probe height change

dutchrazor, 2016-06-03:
Hello,

I have a Cartesian printer on which I set up auto bed levelling. The bed is 200x200mm, for testing it is about 0.5mm non-level. The probe is a Misumi contact switch driven by a small Maxon motor.

First I tried single probes in the same location, this repeats very well. Afterwards I probed the bed via G32 S2 , and the nozzle seems to follow the non-level bed OK. During printing, I had to babystep about -0.3mm to compensate for the inaccuracy measurement of the Z_PROBE_HEIGHT. 

I would then expect to change Z_PROBE_HEIGHT to (Z_PROBE_HEIGHT - 0.3mm). Upon the next print however, the nozzle distance to the bed actually increased and I had to babystep -2.3mm! 

The odd thing is that the levelling itself seems to work, only the Z length is off. I corrected the z-length value manually and now the nozzle follows the bed OK.

Any thoughts on this issue?

// #################### Z-Probing #####################

#define Z_PROBE_Z_OFFSET 0
#define Z_PROBE_Z_OFFSET_MODE 0
#define UI_BED_COATING 1
#define FEATURE_Z_PROBE 1
#define Z_PROBE_BED_DISTANCE 10
#define Z_PROBE_PIN ORIG_X_MAX_PIN
#define Z_PROBE_PULLUP 0
#define Z_PROBE_ON_HIGH 1
#define Z_PROBE_X_OFFSET 24
#define Z_PROBE_Y_OFFSET 34
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 2
#define Z_PROBE_REPETITIONS 3
#define Z_PROBE_HEIGHT 4.35
#define Z_PROBE_START_SCRIPT "M42 P5 S255 \n M42 P6 S170 \n G4 P3000 \n M42 P6 S100"
#define Z_PROBE_FINISHED_SCRIPT "M42 P5 S0 \n M42 P6 S170 \n G4 P3000 \n M42 P6 S0"
#define Z_PROBE_REQUIRES_HEATING 0
#define Z_PROBE_MIN_TEMPERATURE 150
#define FEATURE_AUTOLEVEL 1
#define Z_PROBE_X1 40
#define Z_PROBE_Y1 40
#define Z_PROBE_X2 180
#define Z_PROBE_Y2 40
#define Z_PROBE_X3 110
#define Z_PROBE_Y3 160
#define BED_LEVELING_METHOD 0
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5
#define BED_MOTOR_1_X 0
#define BED_MOTOR_1_Y 0
#define BED_MOTOR_2_X 200
#define BED_MOTOR_2_Y 0
#define BED_MOTOR_3_X 100
#define BED_MOTOR_3_Y 200
#define BENDING_CORRECTION_A 0
#define BENDING_CORRECTION_B 0
#define BENDING_CORRECTION_C 0
#define FEATURE_AXISCOMP 0
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

repetier, 2016-06-04:
That is totally ok. Babystepping does only change z position without telling anyone, so it is not expected that it changes z length or corrects z probe height. You may of course use that information to update z probe height and recalibrate. But it is not intendet to correct that error - it even works without z probe. Main meaning is to eliminate adhesion prints in the beginning.

dutchrazor, 2016-06-04:
Yes, but when I change the Z_PROBE_HEIGHT with the baby stepping difference, I get a totally different bed offset. 

The offset is so large, it it's quite visual. As I said, a 0.3mm difference turned into a 2.3mm difference.

So to recap:

-Initial probing with G32 S2
-Bed is 0.3mm too far from nozzle ( determined with baby stepping)
-Change Z_PROBE_HEIGHT by 0.3mm in EEPROM
-Probe again with G32 S2
-Bed is now 2.3mm too far from nozzle

repetier, 2016-06-04:
Ok, then I misunderstood you. Changing 0.3 should give error 0 or 0.6 depending on direction of correction. That is strange as z probe height just gets added to results from measuring.

Will see if I can reproduce this somehow.

dutchrazor, 2016-06-04:
Thanks, I'll make  copy of the log output this weekend so you have some specific data.

repetier, 2016-06-04:
I just took a look into the code and the z probe height is only added after the repetitions, so I can not see why this happen to you. Since you already plan to make a log, which is a good idea to finding the problem, could you do it twice with exactly the same procedure after homing. The second time with probe height 1 mm less. After homing send
M114
G30 P3

I think that should allow analysing what really happens if I then can compare the 2 logs. Gladly the firmware is quite chatty during probing.

flenger1, 2016-06-21:
Hello,
Printer H-Bot
I have a similar problem, after Z_PROBE (G32 S2/S0) incorrectly determined the length of the axis Z about +- 1mm non-level (random number within this range). Drive z_probe through servo.
 I thought that this mechanical error, the blame endstop axis Z, I replaced them on the hall sensor (z_max and z_probe)-problem persists, replace the drive axis Z on a ball screwr - problem persists...

// #################### Z-Probing #####################

#define Z_PROBE_Z_OFFSET 0
#define Z_PROBE_Z_OFFSET_MODE 0
#define UI_BED_COATING 1
#define FEATURE_Z_PROBE 1
#define Z_PROBE_BED_DISTANCE 5
#define Z_PROBE_PIN ORIG_Z_MIN_PIN
#define Z_PROBE_PULLUP 0
#define Z_PROBE_ON_HIGH 0
#define Z_PROBE_X_OFFSET -6
#define Z_PROBE_Y_OFFSET 21
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 1
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_HEIGHT 8.51
#define Z_PROBE_START_SCRIPT "M340 P0 S2500"
#define Z_PROBE_FINISHED_SCRIPT "M340 P0 S500"
#define Z_PROBE_REQUIRES_HEATING 0
#define Z_PROBE_MIN_TEMPERATURE 150
#define FEATURE_AUTOLEVEL 1
#define Z_PROBE_X1 10
#define Z_PROBE_Y1 5
#define Z_PROBE_X2 170
#define Z_PROBE_Y2 5
#define Z_PROBE_X3 10
#define Z_PROBE_Y3 170
#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
#define BED_LEVELING_GRID_SIZE 3
#define BED_LEVELING_REPETITIONS 5
#define BED_MOTOR_1_X 0
#define BED_MOTOR_1_Y 0
#define BED_MOTOR_2_X 200
#define BED_MOTOR_2_Y 0
#define BED_MOTOR_3_X 100
#define BED_MOTOR_3_Y 200
#define BENDING_CORRECTION_A 0
#define BENDING_CORRECTION_B 0
#define BENDING_CORRECTION_C 0
#define FEATURE_AXISCOMP 1
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

roginvs, 2016-06-21:
Take a look on this #540 , I suppose you have same issue.
Good way (as for me) to measure Z_PROBE_HEIGHT is to:
1. Home XYZ
2. Find a place when nozzle almost touching bed with G1 command (Let it be X=100, Y=100, Z=4.6)
3. Move z axis up for some pre-defined value (like +20 == G1 Z24.6) to be able to safely make measurement
4. Make a measurement with G30 command. Then, you have to tune Z_PROBE_HEIGHT to make G30 command result to be equal to 20 (value added in step 3)
[Have a look at pic](https://hsto.org/files/c97/be6/32f/c97be632f21043a9b1db056e3858136f.png)

dutchrazor, 2016-06-28:
Hey thanks for the tip, I will try that. I didn't manage to get logging yet, but it seems roginvs found the problem in code already.

repetier, 2016-06-29:
Use latest dev version, which has the fix included.

paulnd, 2016-10-24:
Hello, I'm using the latest dev version but also experience this on a cartesian printer. 

My steps are
Measure z probe height by measuring the distance between the head just touching the bed and the height at which the probe triggers and enter into dev configuration web page along with other parameters and settings 
Clear eeprom by uploading a standard eeprom clear sketch 
Upload configuration generated by the dev web page 
Home xyz (Z is max at the top of the printer) 
G30 S2 and the board is probed 

However when I print, the print head is  a few millimetres above the bed and I have to manually adjust the max z height in eeprom until the head is at the correct distance from the bed

Paul