repetier/RepetierFirmware #536: G30 Error:Z-probe failed - Z has no more control

luc-github, 2016-06-04:
Hello,
Using 092.9 FW on cartesian printer, when there is some Z-probe error, (like no probe-response because not right position) the bed does not stop going down after error is raised, I have to do emergency stop as I am afraid to damage printer because I do not have Z max end stop.

Is that an expected behavior ? 

for example I type  __G28__ then __G30 P1__
and here the log. 

```
16:35:33.043 : ok 42
16:35:33.052 : T:22.43 / 0 B:26.06 / 0 B@:0 @:0 T0:22.43 / 0 @0:0 T1:23.78 / 0 @1:0
16:35:33.764 : N43 G30 P1*108
16:35:33.764 : ok 43
16:35:33.832 : Error:Z-probe failed
16:35:35.766 : busy:processing
16:35:36.105 : N44 M105*23
16:35:37.766 : busy:processing
16:35:39.765 : busy:processing
16:35:41.765 : busy:processing
16:35:43.765 : busy:processing
16:35:45.765 : busy:processing
16:35:47.766 : busy:processing
16:35:49.766 : busy:processing
.....
```
the bed going down and seems to willing to stop so I did power off the printer 

Should not bed stop going down after at maximum Z_PROBE_BED_DISTANCE when error happen ?

repetier, 2016-06-04:
There is only one position in code that issues the error:
```
        probeDepth = 2 * (Printer::zMaxSteps - Printer::zMinSteps); // probe should always hit within this distance
        stepsRemainingAtZHit = -1; // Marker that we did not hit z probe
        //int32_t offx = axisStepsPerMM[X_AXIS] * EEPROM::zProbeXOffset();
        //int32_t offy = axisStepsPerMM[Y_AXIS] * EEPROM::zProbeYOffset();
        //PrintLine::moveRelativeDistanceInSteps(-offx,-offy,0,0,EEPROM::zProbeXYSpeed(),true,true);
        setZProbingActive(true);
        PrintLine::moveRelativeDistanceInSteps(0, 0, -probeDepth, 0, EEPROM::zProbeSpeed(), true, true);
        if(stepsRemainingAtZHit < 0) {
            Com::printErrorFLN(Com::tZProbeFailed);
            return ILLEGAL_Z_PROBE;
        }

```
As you see the move down is 2 * z height, so if z probe never triggers the move would be very low. If the message is omitted the function immediately stops.

I'm a bit confused as you say don't stop going down together with max z endstop. So you mean the function finishes and after that it goes up? It's always confusing as some printers move bed others extruder so terminology is not always clear. But with P1 there is no reason to do anything after probing and if it does not work it should already have crashed into the bed BEFORE you get a error message.

luc-github, 2016-06-04:
Sorry for the wording confusion.
probing start, bed go up to find the probe but failed, after error raised the bed go down, slowly but continuously and after 10cm I power off printer because I am afraid it won't stop.

May be it is a corner case, but for this situation G30 P1 is done when probe is out of the bed, probe never get contact with bed, which can happen.

repetier, 2016-06-04:
So you mean it's the z min endstop that stops the bed? You know the problems with z min endstop and z probe I guess.

luc-github, 2016-06-04:
Yes - sorry I may miss the issue - I saw (Z-Probing and X, Y Endstop bug) but was not seems same problem

repetier, 2016-06-04:
Checking the code I can not see the real problem, but I see the current G30 code is different from what you showed in log, so it seems that at least something has already changed in dev.

Could you please tryout dev version for the issue.

luc-github, 2016-06-04:
Ok thanks - I will try dev branch - I was always using work92 until now

repetier, 2016-06-05:
Ok, I switched from work092 to dev to get cleaner naming in future. Now I have only master and dev being active.

luc-github, 2016-06-05:
I forgot to mention there is a typo error in G30 error message it says __G29 leveling failed!__

Anyway I still have issue once I sync with dev branch, if I change probeDepth  to 10 mm 
```probeDepth = static_cast<int32_t>(axisStepsPerMM[Z_AXIS] * 10);```
bed go down 20 mm and stop and got the error __G29 leveling failed!__ and go to dry run mode which I did not had  with previous probeDepth 

After some tests I found that fatal error function is generating the behavior of moving too much the bed so I commented it and print position instead :
```
case 30: 
			{ // G30 single probe set Z0
                uint8_t p = (com->hasP() ? (uint8_t)com->P : 3);
                if(Printer::runZProbe(p & 1,p & 2) == ILLEGAL_Z_PROBE) {
					//GCode::fatalError(PSTR("G30 leveling failed!"));
					printCurrentPosition(PSTR("G30 "));
					break;
				}
                Printer::updateCurrentPosition(p & 1);
            }
            break;
```
here the output :
```
Error:Z-probe failed
X:0.00 Y:0.00 Z:-400.00 E:0.0000
```
obviously the position is not correct, should be Z :0.000 as I am in home position, but it is in -probeDepth  that is why the GCode::fatalError function is moving the bed too low

luc-github, 2016-06-05:
to solve issue I found in __void GCode::fatalError(FSTRINGPARAM(message))__

need to change
```
PrintLine::moveRelativeDistanceInStepsReal(0,0,10*Printer::axisStepsPerMM[Z_AXIS],0,Printer::homingFeedrate[Z_AXIS],true,true);
```
by
```
PrintLine::moveRelativeDistanceInSteps(0,0,10*Printer::axisStepsPerMM[Z_AXIS],0,Printer::homingFeedrate[Z_AXIS],true,true);
```

and no more over move I hope it help

repetier, 2016-06-06:
I like that. In theory both do the same anyway, only real uses lastCmdPos and that is polluted from the error, so your solution should aways work. And it explains why I did not see it in autoleveling code.

closed 2016-06-06 by repetier