repetier/RepetierFirmware #541: more axis for cnc mode

taalabash, 2016-06-25:
Hello there
first I want to thank you for this amazing firmware and especially for the new All-in-one machines support
but is it possible to add more movement axis to this firmware ?? 
I can modify gcode generated from any 4 or 5 axis cam software to replace for example every A axis with E0 and its easy to convert steps per mm to steps per degree , but the problem is that extruders dont have endstops and so i cant use E axis as a movement axis because i cant home this axis 
so is that possible to add endstops to E0 , E1 axis ??

repetier, 2016-06-25:
First understand that the cnc mode was always meant for hybrid printers being able to switch extruders with a mill, so only 3 axis are required.

E0/E1 is all the same. You can only have one active extruder so at max we have 4 axis.

You could add extruder endstop tests and add a e homing predure, then you could use at least 4 axis. Adding more axis is quite straight forward, but needs many places changed and normal speed would degrade as unused axis are still computed or tested if used and both takes time. That and memory consumption (especially in deltas) are the main reasons we never added more axis. And of course we never head the need:-)

kyrreaa, 2016-06-25:
You forget mixing extruder...
That’s not 1 axis anymore, or atleast it’s 3 linked axis.

Kyrre

From: repetier 
Sent: Saturday, June 25, 2016 2:30 PM
To: repetier/Repetier-Firmware 
Subject: Re: [repetier/Repetier-Firmware] more axis for cnc mode (#541)

First understand that the cnc mode was always meant for hybrid printers being able to switch extruders with a mill, so only 3 axis are required.

E0/E1 is all the same. You can only have one active extruder so at max we have 4 axis.

You could add extruder endstop tests and add a e homing predure, then you could use at least 4 axis. Adding more axis is quite straight forward, but needs many places changed and normal speed would degrade as unused axis are still computed or tested if used and both takes time. That and memory consumption (especially in deltas) are the main reasons we never added more axis. And of course we never head the need:-)

repetier, 2016-06-26:
Mixing extruder is still handled as one axis, it just switches the driver getting steps. That does not help here.