repetier/RepetierFirmware #546: Hanging in M109, heating extruders, when extruder already hot

dewhisna, 2016-07-02:
I'm seeing an issue where if the extruder is already hot and an M109 is received, like "M109 T0 S230" to wait for it to heat, it will get stuck waiting for the extruder to get hot even though it already is.  This happens to me regularly since I tend to manually turn on the extruder and run a little filament through to prime it, then pull off the excess filament and start the print.  The only way I can get around it is to turn off the heater, let the temp drop about 5-15 degrees, and start it running so that it hits a "sweet spot" in the temperature rise to skip the wait loop.

And for the life of me, I can't see how the following code snippet in Extruder::setTemperatureForExtruder() in Extruder.cpp ever exits, which I think is key to the problem:
```c++
        do
        {
            previousMillisCmd = currentTime = HAL::timeInMilliseconds();
            if( (currentTime - printedTime) > 1000 )   //Print Temp Reading every 1 second while heating up.
            {
                Commands::printTemperatures();
                printedTime = currentTime;
            }
            Commands::checkForPeriodicalActions(true);
			GCode::keepAlive(WaitHeater);
            //gcode_read_serial();
#if RETRACT_DURING_HEATUP
            if (actExtruder == Extruder::current && actExtruder->waitRetractUnits > 0 && !retracted && dirRising && actExtruder->tempControl.currentTemperatureC > actExtruder->waitRetractTemperature)
            {
                PrintLine::moveRelativeDistanceInSteps(0, 0, 0, -actExtruder->waitRetractUnits * Printer::axisStepsPerMM[E_AXIS], actExtruder->maxFeedrate / 4, false, false);
                retracted = 1;
            }
#endif
            if((waituntil == 0 &&
                    (dirRising ? actExtruder->tempControl.currentTemperatureC >= actExtruder->tempControl.targetTemperatureC - 1
                     : actExtruder->tempControl.currentTemperatureC <= actExtruder->tempControl.targetTemperatureC + 1))
#if defined(TEMP_HYSTERESIS) && TEMP_HYSTERESIS>=1
                    || (waituntil != 0 && (abs(actExtruder->tempControl.currentTemperatureC - actExtruder->tempControl.targetTemperatureC)) > TEMP_HYSTERESIS)
#endif
              )
            {
                waituntil = currentTime + 1000UL*(millis_t)actExtruder->watchPeriod; // now wait for temp. to stabilize
            }
        }
        while(waituntil == 0 || (waituntil != 0 && (millis_t)(waituntil - currentTime) < 2000000000UL));
```

If I am reading that code correctly, it can only set the `waituntil` time once (since in my case `TEMP_HYSTERESIS` is set to 0), and `dirRising` never changes inside the loop.  And then the loop runs while `waituntil` isn't set (is 0) or has a delta with `currentTime` less than something just over 23 days.  But what gets it out of that loop when it actually gets hot, without waiting out those 23 days?

The only way I can see it not getting stuck is if the initial check of:
```c++
    if(wait && temperatureInCelsius > MAX_ROOM_TEMPERATURE
#if defined(SKIP_M109_IF_WITHIN) && SKIP_M109_IF_WITHIN > 0
            && !(abs(tc->currentTemperatureC - tc->targetTemperatureC) < (SKIP_M109_IF_WITHIN))// Already in range
#endif
      )
```

keeps it out of the loop in the first place -- i.e. the heater is within the `SKIP_M109_IF_WITHIN` range or if the heater is still at room temperature (which doesn't make sense why it would skip the wait if it's still at room temp, shouldn't that be the other way around?  or why does it even care about room temp at all?) .

What am I missing?  How does it get out of that loop?  Why does it get stuck heating?  And why do I not see any other issue reports with people having similar problems?

The full gcode for the init process, in case it matters, is similar to the following:
```gcode
G28 ; Home extruder
G1 Z15 F100
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
M190 S110
; Activate all used extruder
M104 T0 S230
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
M109 T0 S230
;Layer count: 125
;LAYER:0
M107
G0 F3600 X71.754 Y71.324 Z0.200
```

Thanks!

repetier, 2016-07-02:
Ok, first inital check is clear - if you are with SKIP_M109_IF_WITHIN you are done. You aren't so you apparently go into main loop. That exists is waituntil < currentTime neglecting the special handling of first loop and overflow.

My guess is your TEMP_HYSTERESIS > 0 so everytime your temperature swings out of target range the time gets reset. You must be watchPeriod ms inside the target range. Alternatively check what your watchPeriod is, but since it works from cooled down I don't think it is that high. Last possible reason is you swing not enough so the dirRising condition never is met. Some settings cause temperature constantly being below set temp. and only the first overswing takes them over the required value.

dewhisna, 2016-07-02:
No, TEMP_HYSTERESIS is 0.  Here's the snippet from the Configuration:

```c++
#define RETRACT_DURING_HEATUP true
#define PID_CONTROL_RANGE 60
#define SKIP_M109_IF_WITHIN 2
#define SCALE_PID_TO_MAX 0
#define TEMP_HYSTERESIS 0
#define EXTRUDE_MAXLENGTH 160
```

And actually, it's the opposite of your hypothesis.  It's always above target.  It's setup to use deadband control and with 230 dialed in as the set point or target, the temperature on the LCD bounces between 232 and 236 or so for actual temp.  The value reported back to the computer seems to always be 232 or 233, which I attributed to different sample times between the two data outputs.

So in actuality, my temperature never goes below the target.

If the extruder is already hot when it gets to this function, then `dirRising` could actually indicate that the temperature is falling instead of rising, if it was currently above the target at the time, and outside of the "skip within range".  Or it could even be above target and still rising.  Plus that `dirRising` variable never gets updated inside the loop -- whatever it is at the start of the function is it (shouldn't it get moved to inside the 'while' loop so that `dirRising` remains correct?).

Now, I'm theorizing that if I increase the `SKIP_M109_IF_WITHIN` value to 10 or 15 or something, things would work better and likely work around my problem, since that's outside the control window of the deadband control.   But regardless, that does mean that this function can easily get stuck waiting for it to get hot with certain setting values, and would seem to be a point of concern.

Since the primary function of this loop is to wait for things to heat up, then I would think that adding a simple check of `tc->currentTemperatureC >= tc->targetTemperatureC` to the while loop would be a sufficient fix to keep from ever getting stuck in the loop.

repetier, 2016-07-02:
Ok, with your new data it is clear what happens. dirRising false, so you must satisfy
actExtruder->tempControl.currentTemperatureC <= actExtruder->tempControl.targetTemperatureC + 1
so with your 230°C example you would need to be below 231°C but your swing between 232 and 236 so that condition is never met. Putting dir inside loop would also not help since you stay above set temperature and loop waits for a cooldown that never happens.

So from logical view it is all ok and working as expected, only that your dead time setting has pwm set too high or timing to long I think, so it powers to much/early keeping you above the target and not around the target temperature. You could simply increase the +1 to + SKIP_M109_IF_WITHIN to relax the condition and allowing exit on a broader range or adjust dead time parameter.

dewhisna, 2016-07-02:
OK, that makes sense.  Though if relaxing the +1 to +SKIP_M109_IF_WITHIN value, I would think for consistency in this algorithm it would good to change the other side too so that it reads:

```c++
(dirRising ? actExtruder->tempControl.currentTemperatureC >= actExtruder->tempControl.targetTemperatureC - SKIP_M109_IF_WITHIN
    : actExtruder->tempControl.currentTemperatureC <= actExtruder->tempControl.targetTemperatureC + SKIP_M109_IF_WITHIN))
```

And I agree with your assessment of the dead time setting.  I'm sure it's off.  I was starting with the initial values used for my printer (the BoXZY) which was originally setup for the 0.91.7 firmware.  I've been working on pulling it up to the latest 0.92.9 version and addressing problems like this.

Long term, I'm planning to switch it to a PID loop and use the auto-tune function (hopefully) to zero in the gains.  It will be interesting to see what it comes up with for gains for various fan speed combinations, as one of the original BoXZY firmware customizations was to add a hack around over cooling of the extruders due to the fan.  Their fan design is really neither an extruder fan nor a print cooling fan (it's sort of both and neither).  So when running the fan to cool the print, you inadvertently affect temperature control and make that unstable.

Their hack was originally helping the overshoot issue with the bad dead time setting and compensating for the over cooling from the fan.  But, I had to remove their hack because it was triggering the "thermistor decoupled" fault detection logic on the newer firmware.  I'm hoping that switching things to PID and getting it properly calibrated will negate the need for their hack entirely.

Until I can work through those details, I think I'll relax this check as you suggest, and bump SKIP_M109_IF_WITHIN a few degrees (perhaps 4 would be a good number?).

I still think it would be good to add some sort of safeguard on this function to help avoid it getting stuck in general when people have slightly miscalibrated firmware settings.  It would be a good enhancement.  You can decide how you wish to address that and close this ticket as appropriate.  I think I now understand what is going on and how to solve it for this printer.

Thanks!

repetier, 2016-07-08:
New development release has now a relaxed condition. If you are within 5°C a 2 minute countdown starts that will exit if you do not get temperature inside the set corridor. That should cover normal misconfiguration issues and gives enough time to exit on the narrow corridor as planned.

closed 2016-07-08 by repetier