repetier/RepetierFirmware #551: bug in trapezoid planning

gr5, 2016-07-18:
I spent about 40 hours looking through the motion planners of repetier, Marlin, Redeem.  Repetier has a bug that made it's way to Redeem but is not in Marlin.  The math is a bit complex and I like to think visually but I don't know who wrote the code so I'm not sure if they think visually (probably) but I can see exactly one way they may have made the error.  Anyway the error is in motion - when it's calculating the steps for the trapezoids if it reaches the full speed (feedrate from gcodes) all is good but if there is no plateau on the trapezoid (again visually) then you need to find the intersection where acceleration and deceleration intersect.  Here's the code in question - line 901 of motion.cpp

```
    if(accelSteps + decelSteps >= stepsRemaining)   // can't reach limit speed
    {
        uint16_t red = (accelSteps + decelSteps + 2 - stepsRemaining) >> 1;
        accelSteps = accelSteps - RMath::min(accelSteps, red);
        decelSteps = decelSteps - RMath::min(decelSteps, red);
    }
```

"red" I assume stands for reduce?  It's the overlap of the acceleration and deceleration phases.  In distance.  However when thinking about the trapezoids it is only shaped like a trapezoid in the time domain, not the distance domain.  The whole "trapezoid" concept comes from when you plot velocity on the vertical axis and time on the horizontal axis.  In this case since acceleration can only be 3 values (zero, or plus and minus default acceleration) then you get trapezoid graphs.  But in the distance domain you get parabolic curves.  The above code works fine if the start and end velocities are equal but that's often not the case.  So you will get wrong (very wrong) values for accelSteps and decelSteps.  Either too large or too small.  Why did no one notice this before?  Maybe they did.  The problem only occurs on short segments (<3mm) and higher speeds (>40mm/sec)  and where "jerk" limits vertices to *different* speeds (which is probably rare).  You see short segments on curves but with circular curves you get consistent jerk limitations at each vertex.  But with a more complex curve you would get this issue.  What will happen?  You may exceed the max acceleration parameters (or max instantaneous change in velocity) at vertexes.  You could lose steps and see the part "lean".  Slowing the print down would hide the problem.

How does Marlin get it right?  They took 3 equations with 3 unknowns and did massive amounts of algebra to come up with this relatively simple looking formula.  I tried working through it - what a nightmare of algebra - but the formula is correct:

```
FORCE_INLINE float estimate_acceleration_distance(float initial_rate, float target_rate, float acceleration)
{
  if (acceleration!=0) {
    return((target_rate*target_rate-initial_rate*initial_rate)/
      (2.0*acceleration));
  }
  else {
    return 0.0;  // acceleration was 0, set acceleration distance to 0
  }
}
```

I can post more photos and descriptions of what is wrong with the "red" concept above and why I suspect the programmer did what s/he did.

gr5, 2016-07-18:
Oh hell.  I posted the wrong marlin snippet.  Here is the correct one (sloppy cut/paste mistake):

```
FORCE_INLINE float intersection_distance(float initial_rate, float final_rate, float acceleration, float distance) 
{
  if (acceleration!=0) {
    return((2.0*acceleration*distance-initial_rate*initial_rate+final_rate*final_rate)/
      (4.0*acceleration) );
  }
  else {
    return 0.0;  // acceleration was 0, set intersection distance to 0
  }
}
```

repetier, 2016-07-24:
Sorry for delay, but I was on holiday.

Interesting thought, also I can do not come to your conclusion for now. You are right that only speed-time is trapezoid, but our solution takes advantage of the symmetry when reaching the final speed. You take away the same part for acceleration and deceleration so it keeps balanced. I made a small excel sheet to compare Marlin and Repetier solution using floating math, so neglecting the +1 added in repetier for safety.
[accelstepscomp.xlsx](https://github.com/repetier/Repetier-Firmware/files/380195/accelstepscomp.xlsx)
As you see I always get the same solution with a much simpler formula. So in general it should have the same result, but maybe I overlooked one condition giving different results. From pre planning it should always be possible to go from speed 1 to speed 2 so conditions where this is not possible should never appear.

One problem with high resolution printers might be the use of uint16 for red/accel/decel to reduce memory usage and increase speed. acceleration to max speed must be possible within 32700 steps or you might get a overflow. With typical accelerations no problem.

So yes, please provide the example where repetier result is wrong.

gr5, 2016-07-28:
Rats I screwed up. You are correct.  I think visually.  What I failed to realize is that even though the curves are parabolic the green arrowed sections are 100% symmetrical.  When I drew this out the first time I didn't draw them symmetrical.  Here is your example (80mm start speed, 20mm end speed, 1200mm/sec/sec acceleration, 100mm/sec goal speed, 3mm travel).  Horizontal axis is in mm and is 1mm per division.  Vertical axis is velocity.  "redX2" is shown graphically.  The two "red" sections are symmetrical so you can subtract either from the accel or decel curve to find the intersection point.  Because the accel/decel is identical (1200mm/sec/sec) the "red" curves are always going to be mirror image from the intersection speed up to the goal speed.  I'll close this issue.

[image]

closed 2016-07-28 by gr5