repetier/RepetierFirmware #553: 12864 smart controller

5stringmonster, 2016-07-20:
Doesn't seem to be fully working. Plug in the printer and it beeps once, screen flashes blue for a second and then nothing else comes up on the screen. The rotary encoder seems to get a response from the ramps 1.4 board, the light flashes on the controller, but screen stays dark. Used the online config tool .92 and picked the right one for it.

sanchosk, 2016-12-11:
This might NOT be the problem of the firmware.
Please, check internet threads about the chinese defective boards - the contrast on the boards is so weak you can't even see anything. Once you add ~230 ohm resistor between pins 2 and 3 on the board (google some how-tos), it's working OK.

repetier, 2016-12-12:
Light is not controlled by the firmware. Light may get darker when heaters/motors get turned on as the voltage may drop. Flickering with no load is a hardware problem.

sanchosk, 2016-12-12:
Considering how old the report is (I noticed only now), I'd suggest to close this issue.
I am sure it was a HW problem [like this](http://forums.reprap.org/read.php?13,448609,448773#msg-448773).

closed 2016-12-12 by repetier