repetier/RepetierFirmware #556: Compilation error (configured online)

Jerimon1998, 2016-07-29:
This is the download from the configuration I created based off my printer's old firmware (copied over the settings that were in common, and measured those that were not):
[Repetier-Firmware-2016-07-25.zip](https://github.com/repetier/Repetier-Firmware/files/391193/Repetier-Firmware-2016-07-25.zip)
The rest of the details for this issue is at the[ repetier software forums](https://forum.repetier.com/discussion/2347/compilation-error-eeprom-cpp-659-error-expected-primary-expression-before-token).

jamesarm97, 2016-07-29:
do you have a define like the following in your configuration.h file?

#define Z_PROBE_X_OFFSET 0
#define Z_PROBE_Y_OFFSET 0

Jerimon1998, 2016-07-30:
In the past 24 hours I have been working to solve most of the problems I had. Most of them turned out to be user error (I messed something up and didn't realize/understand it). I have finally moved on to actually uploading code to the board, but seem to be having a problem doing so. The USBasp that I am using (and now have correctly connected) seems to be "stuck" during programming. The action LED stays solid, as if it is still programming, even after the Arduino IDE is no longer showing that it is uploading. I am using the "Upload using programmer" method, with USBasp selected in the tools menu, and the corresponding drivers installed. The IDE acts as if everything is fine, and even told me when I had the connector oriented incorrectly (as I later learned).
In other words, there is no problem with the latest firmware download (based off the entry of the missing value that caused my first problem) as far as compiling goes. All I need now is for the firmware to make its way into the board correctly.

Jerimon1998, 2016-07-30:
I have finally deduced that the culprit is the programmer. Therefore, I am now closing this issue. I thank jamesarm97 and anyone who looked at this issue for their time. There are two main points that I have to make from this: Firstly, the configuration wizard could be improved by forcing any value left blank to its default. This would remove the issue of missing values when compiled. **More importantly, there was no "Repetier.cbp" or "Repetier.cpp" file to be found in the download. The .ino file did not work in the codeblocks program, even when the the file ending was changed. At the end of compilation, it would always come up with "Error linking objects."**; or, if tried to compile duplicate files (as it tends to do on second runs), it would come up with a slurry of errors.

closed 2016-07-30 by Jerimon1998

repetier, 2016-07-31:
Official codeblocks support has been removed. Problem was that Arduino could not compile any more (compile behaviour ha schanged) when codeblocks had it's extra files includes.