repetier/RepetierFirmware #559: Azteeg X3 with Viki LCD - Error Compiling Arduino 1.6.9

dvseliteus, 2016-08-03:
This is my first time trying the Repetier Firmware and I am trying to program an Azteeg X3 with Viki LCD.
I am receiving the following error during verification: [http://pastebin.com/MgdA5Hee](url)
I used the Repetier-Firmware configuration tool for version 0.92.9.
Are there some settings I have done incorrectly?
Here is my configuration.h file: [http://pastebin.com/aLkxsPsm](url)

repetier, 2016-08-04:
There seems to be no real error. Just the linker exiting with error code but without error message. Searching for this shows that it is more a linker problem. Some users succeeded by simply replacing ld.exe with a other version (or using other arduino ide version). You could also upgrade to dev release where you should get less warnings and maybe the ld error does not happen. I don't think you did anything wrong.

dvseliteus, 2016-08-04:
Thank you for the reply that solved the issue and is working now.

closed 2016-08-04 by dvseliteus