repetier/RepetierFirmware #560: Setting up thermocouple (K type)

rcboats1, 2016-08-04:
II am trying to use a K type thermocouple and am having no luck with getting correct readings? I am using and Azteeg X3 pro board and have set Temperature sensor as AD595 as recommended by Roy at Panucatt, but no luck. I have tried several different sensor type settings and can't find what works, anyone have an idea?

rcarlyle, 2016-08-04:
Are you using the TC1/TC2 inputs? If Roy says AD595, then that's the one
you should be using, there's no point in trying other sensor types.

What readings ARE you getting?

rcboats1, 2016-08-04:
Yes, TC1 and pin 4. I get a reading of 0 deg.

rcboats1, 2016-08-04:
#define EXTRUDER_FAN_COOL_TEMP 50
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_STEPS_PER_MM 97
#define EXT0_TEMPSENSOR_TYPE -1
#define EXT0_TEMPSENSOR_PIN TEMP_4_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
#define EXT0_STEP_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE 1
#define EXT0_ENABLE_PIN E0_ENABLE_PIN
#define EXT0_ENABLE_ON 0
#define EXT0_MAX_FEEDRATE 100
#define EXT0_MAX_START_FEEDRATE 20
#define EXT0_MAX_ACCELERATION 10000
#define EXT0_HEAT_MANAGER 3
#define EXT0_WATCHPERIOD 1
#define EXT0_PID_INTEGRAL_DRIVE_MAX 255
#define EXT0_PID_INTEGRAL_DRIVE_MIN 40
#define EXT0_PID_P 7
#define EXT0_PID_I 2
#define EXT0_PID_D 40
#define EXT0_PID_MAX 255
#define EXT0_ADVANCE_K 0
#define EXT0_ADVANCE_L 0
#define EXT0_ADVANCE_BACKLASH_STEPS 0
#define EXT0_WAIT_RETRACT_TEMP 150
#define EXT0_WAIT_RETRACT_UNITS 0
#define EXT0_SELECT_COMMANDS ""
#define EXT0_DESELECT_COMMANDS ""
#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN_PIN
#define EXT0_EXTRUDER_COOLER_SPEED 255

rcarlyle, 2016-08-04:
Are you sure that's the right pin?

rcboats1, 2016-08-04:
Yes, it's analog pin 4.

I have it working now. I had to use AD8495 for the sensor type. 

Thank you.

repetier, 2016-08-05:
TEMP_4_PIN is analog 11!
You need thermistor inputs
#define THERMOCOUPLE_0_PIN         4  
#define THERMOCOUPLE_1_PIN         5  
or just set explicitly 4.