repetier/RepetierFirmware #561: FEATURE_BEEPER breaks compilation

josemariaaraujo, 2016-08-05:
If I enable FEATURE_BEEPER in Configuration.h I get a compilation error in the Arduino IDE. I'm using Repetier Version 0.92.9, and I attach my Configuration.h. It compiles as is, but change `#define FEATURE_BEEPER 0` to 1 to get the error.

The error I get is:
`'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

'stepperWait' referenced in section '.text' of C:\Users\jm_ar\AppData\Local\Temp\ccgZu3VY.ltrans9.ltrans.o: defined in discarded section '.text' of sketch\HAL.cpp.o (symbol from plugin)

collect2.exe: error: ld returned 1 exit status

exit status 1
Erro ao compilar para a placa Arduino/Genuino Mega or Mega 2560.`

repetier, 2016-08-06:
There is no real connection between beeper and stepperWait. My assumption is that you have selected too many languages for the ui, so you get more then 64k of data which is then not addressable anymore and would match to the discarded section message.

josemariaaraujo, 2016-08-06:
The only language selected is English as is shown in the Configuration file included.

zverugaz, 2016-09-22:
Confirm. If I change FEATURE_BEEPER to 1 scketch not compilling.

agonysfate, 2017-02-08:
First time I tried the Web config wizard and it has some errors too
today I try this copy here and Configuration.h is widely different and have same problem when I enable Buzzer it compile error 
the web copy can run buzzer and sounds without problems and the menus are more dense and have more pages than this copy here,
I think I better take the Config from here and Firmware from web, that might fix the problem.
one thing is Im after is to run the firmware on ArduinoDUE as I have that board but dont want to use it yet until I finish all tweaks needed for Mega first

repetier, 2017-02-09:
Github and online config firmware are 100% identical. Only difference is configuration.h which has same options but without comments and with id string to reread it in config tool.