repetier/RepetierFirmware #562: configuration tool: download buttons

piersh, 2016-08-06:
is there any chance you can change the way the download buttons work on the last page of the configuration tool?

currently, when you press "Download Configuration.h" it replaces the page with the contents of Configuration.h, which is great. But the problem is that if you want to (repeatedly) tweak a configuration setting and download another configuration.h you have to start all over from the beggining uploading the configuration.h. this could be improved by either:
1) opening the download in a new window ("_blank", instead of "_self"), in configurator.js or
2) adding a "Content-Disposition: filename='Configuration.h'" header in config.php.

Kulitorum, 2016-09-04:
This does not happen to me, my computer just downloads the .h file as a file, and the webpage stays up.

I think you need to change some preferences in your browser - start by trying to use another browser, and go from there.

Spongman, 2016-12-07:
looks like this was fixed on the server with the adding of a Content-Disposition header

closed 2016-12-07 by Spongman