repetier/RepetierFirmware #563: Print offsets (from LCD menu) not working on deltas

romekb, 2016-08-12:
Hi !
I found that print offsets not working with delta printers. Delta homing cancel all defined offsets.
I comment lines:
`coordinateOffset[X_AXIS] = 0;`
`coordinateOffset[Y_AXIS] = 0;`
`coordinateOffset[Z_AXIS] = 0;`
in `void Printer::homeZAxis() // Delta z homing` and seems to working OK.
I'm not sure for simple comment these lines, maybe we need store offsets before homing and restore it after end of homing, but for now the simple comment work for me.

repetier, 2016-08-13:
Yes, homing is supposed to reset old offsets. It does this also for cartesian printers. You can use G92 afterwards to set any offset you want, but at least then you know what the base is. If it never resets that is hard to say as each G92 is relative to last one.

repetier, 2016-08-13:
The offsets in LCD are the values set be G92 btw. so maybe you expected something different.

closed 2016-08-13 by repetier

romekb, 2016-08-13:
I expected to set offset from LCD menu (for move to outside damaged 
buildtak area) then start any print from SD card.
Without need to slicing again my model at different bed position.
The offsets work perfectly in this case, but now the homing function 
executed at start of SD printing cancel my offsets.

		Pozdrawiam  Romuald Bialy
		E'mail:     romek_b@o2.pl

W dniu 2016-08-13 o 14:20, repetier pisze: