repetier/RepetierFirmware #566: Z homing Problem

JohnsonLv, 2016-08-20:
Dear  Repetier,

I met a problem , I uses Repetier-firmware 0.92 ,the endstop config like below ,but my printer Z axis home randomly and not hint the Z max endstop finish the Z homing ! Please help me solve the issue!
Thank you in advanced !

BR,Johnson 

// ################ Endstop configuration #####################

#define ENDSTOP_PULLUP_X_MIN true
#define ENDSTOP_X_MIN_INVERTING true
#define MIN_HARDWARE_ENDSTOP_X true
#define ENDSTOP_PULLUP_Y_MIN true
#define ENDSTOP_Y_MIN_INVERTING true
#define MIN_HARDWARE_ENDSTOP_Y true
#define ENDSTOP_PULLUP_Z_MIN true
#define ENDSTOP_Z_MIN_INVERTING true
#define MIN_HARDWARE_ENDSTOP_Z false
#define ENDSTOP_PULLUP_X_MAX true
#define ENDSTOP_X_MAX_INVERTING true
#define MAX_HARDWARE_ENDSTOP_X false
#define ENDSTOP_PULLUP_Y_MAX true
#define ENDSTOP_Y_MAX_INVERTING true
#define MAX_HARDWARE_ENDSTOP_Y false
#define ENDSTOP_PULLUP_Z_MAX true
#define ENDSTOP_Z_MAX_INVERTING true
#define MAX_HARDWARE_ENDSTOP_Z true
#define max_software_endstop_r true

#define min_software_endstop_x false
#define min_software_endstop_y false
#define min_software_endstop_z false
#define max_software_endstop_x false
#define max_software_endstop_y false
#define max_software_endstop_z false
#define ENDSTOP_X_BACK_MOVE 5
#define ENDSTOP_Y_BACK_MOVE 5
#define ENDSTOP_Z_BACK_MOVE 2
#define ENDSTOP_X_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Y_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Z_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_X_BACK_ON_HOME 1
#define ENDSTOP_Y_BACK_ON_HOME 1
#define ENDSTOP_Z_BACK_ON_HOME 1
#define ALWAYS_CHECK_ENDSTOPS 1

repetier, 2016-08-20:
First check M119 and see if it returns z max signals. test by triggering by hand.

Your excerpt shows not homing direction nor did you tell us enough details like what is happening when you home.

JohnsonLv, 2016-08-23:
I use M119 the signl test is OK ,but the Z axis not finish homing and still move .

repetier, 2016-08-23:
Could be crosstalk triggering endstop before real hit. Are cables twisted and/or shielded for endstop?
Otherwise give more information about what exactly is happenening so we can imagine at least what the problem is.

Kulitorum, 2016-09-04:
Try to untangle your wires and then avoid putting endstop and stepper motor wires next to eachother. 

Alternavtily pair-twist your stepper wires (twist Red/blue together and Black/green together) to avoid the steppers inducing power into the endstop wires, which is known to cause false endstop triggers.