repetier/RepetierFirmware #568: Illegal homing order for z probe based homing when homing is Y,X then Z?

strips, 2016-08-30:
Hi,

I'm using DEV branch and config tool to set up my first Repetier for BigBox and getting compilation a error.

As I understand it XY needs to be known before homing Z min with Z prob.

I have to home in the order Y, X then Z because I have a dock / dump bucket at Y max. When at the dock I can not move along X as it would collide. 

The following code seems strange and contradicting.

`
#if HOMING_ORDER != HOME_ORDER_XYZ && HOMING_ORDER != HOME_ORDER_XZY && HOMING_ORDER != HOME_ORDER_ZXYTZ && HOMING_ORDER != HOME_ORDER_XYTZ
	#error Illegal homing order for z probe based homing!
`

repetier, 2016-08-30:
Aparrently an error, should be
#if HOMING_ORDER != HOME_ORDER_XYZ && HOMING_ORDER != HOME_ORDER_YXZ && HOMING_ORDER != HOME_ORDER_ZXYTZ && HOMING_ORDER != HOME_ORDER_XYTZ
which would also cover your case. Have fixed it for next update.

closed 2016-08-30 by repetier

strips, 2016-08-30:
Thanks :) Was afraid to fix it manually if there was a reason :)