repetier/RepetierFirmware #569: Autolevel from display?

Kulitorum, 2016-09-02:
I use a RUMBA with the Geeetech Full Graphic LCD. So far I have always used repetier host to run my printers, but I have started using only the display, and it suits me well now that I have more printers then USB ports in my computer :)

But is there anyway to run a G32 S2 command from the display? Annoying to have to connect a computer, just to send a single gcode line.

Thanks for all your hard work

Kulitorum, 2016-09-04:
In fact, now I think about it, it would be cool to have a "Scripts" section on the display, so I could somehow put small custom scripts of gcode and run them at will from the display menu.

repetier, 2016-09-04:
Maybe adding a extra menu "Special Commands" where we could add 10 defined gcodes. Then config would require pairs of command name/gcode which would be included in gcode. We could even add flags to when show the command, e.g. not during print.

Kulitorum, 2016-09-04:
Sounds good. "Gcode" in this context should be allowed to be multi-line though, and not just a single command....?

repetier, 2016-09-04:
All firmware gcodes in strings are multiline separated by \n, so that would be no problem.

Kulitorum, 2016-09-04:
Cool.
I have a complete printer that I need to deliver to my client, any chance you can guide me how to add a menu point right now, so I can finish this printer? - I have done some programming in the past, but not for the last two years, so I may be a little rusty - but with a few pointers, I guess I can make something work.

Or are you going to ditch out a beta version of this on the development branch in a weeks time?

Kulitorum, 2016-09-04:
BTW, your firmware is hands down the best around, and I want to thank you for the huge amount of work you are putting into this. I just donated, and I hope business is going well, and that you won't stop developing this any time soon :)

repetier, 2016-09-04:
No plans to stop developing it.

End of september I will start with a bigger change in ui so I guess then I'll put it there.

Intermediate solution:
1. Add a G32 S2 into a file on sd card.
2. Add a new action + menu entry. In menu action put
case MYACTIONID:
 GCode::executeFString(PSTR("G32 S2"));
break;
and add a menu entry in uimenu.h

luc-github, 2016-09-04:
If I may participate as I already implemented it in Davinci fork - there is one thing you need to handle if you are not connect to host: it is the fatalerror management, it locks everything if an error happen.

I have added in the gcode menu the parameter I (for interactive as it display also the mesured positions and propose to save or not), same for G29 of course 

https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Commands.cpp#L1270

```
 if(!runBedLeveling(com)) {
				//Davinci Specific
				if (!(com->hasI()))GCode::fatalError(PSTR("G32 leveling failed!"));
				else PrintLine::moveRelativeDistanceInSteps(0,0,10*Printer::axisStepsPerMM[Z_AXIS],0,Printer::homingFeedrate[0],true,false);
				Printer::zprobe_ok = false;
				Printer::homeAxis(true, true, true);
				break;
			}
```

Just my 2 cents

Kulitorum, 2016-09-04:
Have a working hack now, thanks for the help :)

Kulitorum

closed 2017-11-05 by repetier