repetier/RepetierFirmware #570: Crash after G30 command (Zonestar P802N Melzi 2.0 V0.92.9)

xpix, 2016-09-03:
Hello Friends,

i recompile the zonestar master of repetier firmware and add my ABL capacitive Sensor. 

If i send the "G30" command, then he try first to home X-Axis and after ~1 second the firmware is freezed. The controller don't receive more commands and i have to reset with the "Emergency Button".

Here my Configuration file:

``

/*
    This file is part of Repetier-Firmware.

    Repetier-Firmware is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Repetier-Firmware is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Repetier-Firmware.  If not, see <http://www.gnu.org/licenses/>.

*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**************** READ FIRST ************************

   This configuration file was created with the configuration tool. For that
   reason, it does not contain the same informations as the original Configuration.h file.
   It misses the comments and unused parts. Open this file file in the config tool
   to see and change the data. You can also upload it to newer/older versions. The system
   will silently add new options, so compilation continues to work.

   This file is optimized for version 0.92
   generator: http://www.repetier.com/firmware/v092/

   If you are in doubt which named functions use which pins on your board, please check the
   pins.h for the used name->pin assignments and your board documentation to verify it is
   as you expect.

*/

#define NUM_EXTRUDER 1
#define MOTHERBOARD 63
#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
// ################ END MANUAL SETTINGS ##########################

#undef FAN_BOARD_PIN
#define FAN_BOARD_PIN -1
#define BOARD_FAN_SPEED 255
#define FAN_THERMO_PIN -1
#define FAN_THERMO_MIN_PWM 128
#define FAN_THERMO_MAX_PWM 255
#define FAN_THERMO_MIN_TEMP 45
#define FAN_THERMO_MAX_TEMP 60
#define FAN_THERMO_THERMISTOR_PIN -1
#define FAN_THERMO_THERMISTOR_TYPE 1

//#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using Arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not being compatible!
//#define COMPAT_PRE1
[... 1021 lines cut ...]
    "bedMotor2Y": 0,
    "bedMotor3X": 100,
    "bedMotor3Y": 200,
    "zProbeRequiresHeating": "0",
    "zProbeMinTemperature": 150,
    "adcKeypadPin": "1",
    "sharedExtruderHeater": "0",
    "extruderSwitchXYSpeed": 100,
    "dualXAxis": "0",
    "boardFanSpeed": 255,
    "hasMAX6675": false,
    "hasMAX31855": false,
    "hasGeneric1": false,
    "hasGeneric2": false,
    "hasGeneric3": false,
    "hasUser0": false,
    "hasUser1": false,
    "hasUser2": false,
    "numExtruder": 1,
    "version": 92.9,
    "primaryPortName": ""
}
========== End configuration string ==========

*/

xpix, 2016-09-03:
Hi, me again. I also try the developent version. Same effect, crash after G30.

repetier, 2016-09-04:
There is no xhoming for G30. Depending on P it will activate probe and probe position which may be a move.

If you have updated z probe after first install have a look into your eeprom values. The one for z probe may therefore be uninitalized. You could also restore config values with
M502
M500

Also check that you have 1kb ram free on connection (see log). For non deltas normally no problem.

closed 2016-09-04 by repetier

xpix, 2016-09-04:
Oh man, thank you for ur fast solution. Yes the EEPROM Values was wrong. No Crash anymore.

Thank you, you made a happy Sunday for me :+1:

xpix, 2016-09-04:
Sorry, the first move was succesful and then i have the same effect. Here my log from host after input M502,M500,G31,G30:

**11:12:38.963 : Z-probe state:L
11:12:48.879 : Z-probe state:H
11:12:54.556 : Printer reset detected - initalizing**
11:12:54.556 : start
11:12:59.433 : Free RAM:12201
11:12:59.433 : SelectExtruder:0
11:13:04.325 : SD init fail
11:13:04.372 : FIRMWARE_NAME:Repetier_0.92.9 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
11:13:04.372 : Printed filament:0.00m Printing time:0 days 0 hours 0 min
11:13:04.372 : PrinterMode:FFF
11:13:04.405 : X:0.00 Y:0.00 Z:0.000 E:0.0000
11:13:04.409 : DebugLevel:6
11:13:04.425 : SelectExtruder:0

repetier, 2016-09-04:
Looks like your board resets. I guess it is the servo. Do you power it separately or from Arduino 5V. Latter will normally not work.

xpix, 2016-09-04:
Hmm, i don't use a servo :) it's a capacity sensor.

xpix, 2016-09-05:
Sorry for my short message yesterday, but i hade only my remote access. 
So, i can wait for a solution, because i use this endstop now for a normal Z_Min Switch, works very well. I'll add some code for the firmware for my [XDisPlace Project](https://github.com/xpix/XDisPlace/wiki) and can try to fix this crash-error myself at this development process.

repetier, 2016-09-05:
I saw using sensvo pin 11 but missed that the feature is disabled.

For finding the crash position write some text to log to see how far you get. Maybe with a delay followed to make sure it gets send before crash like

`Com::printFLN(PSTR("Pos 1"));HAL::delay(50);`

after some iterations you know extactly where it happens. Must be somehwere in runZProbe in BedLeveling.cpp from your description.

xpix, 2016-09-05:
Ok, nice catch .. I'll try to set servo pin to zero, but the servo feature is deactivated. We will see when i add the debug code.

xpix, 2016-09-12:
Ok found the Problem, if we set:

#define Z_PROBE_START_SCRIPT "G28 XY"
#define Z_PROBE_FINISHED_SCRIPT "G0 X100 Y100 Z10"

Then he try to start the start script, but this will freeze or reset the controller. Maybe a newline at the end of this string will fix this, i empty this strings and now he try to measure the 3 points.

repetier, 2016-09-12:
Maybe write it correct like G28 X0 Y0 also I don't understand why you need to home to activate z probe.

xpix, 2016-09-12:
I got this config from another guy he published his configuration file. To home xy sounds normal, cuz RT move to the probe positions and this values are relative to XY Zero.

xpix, 2016-09-12:
Ok, btw. the start script function doesnt work and let the controller freeze everytime. With newline and w/o i got an reset.

repetier, 2016-09-15:
Does it also freeze with a simple command like G4 or only with homing?