repetier/RepetierFirmware #571: "Extr1 z-offset" not used

Kulitorum, 2016-09-04:
My z-probe is offset 30,10,5mm - so it's above the nozzle.

So I use 3-point auto leveling and use 3 spots where the nozzle is outside the bed, but the zprobe is not.

I have defined my "Z-probe-height" at 10,8mm to force the printer to lower the bed enough between probes that the nozzle does not hit the bed, and I was planning to use the "Extr1 z-offset" to make the nozzle hit the heatbed at printing.

Unfortunatly, it seems the  "Extr1 z-offset" is not used. No matter what I put there, -10000 or +10000 the nozzle always prints about 8mm over the bed.

Kulitorum, 2016-09-04:
PS: When I change the "Extr1 z-offset" "live", the bed moves up, and I can tune it to the correct height, but when I home the printer and do "G1 X150 Y150 Z0" the nozzle ends up 8mm over the bed again.

Kulitorum, 2016-09-04:
Hmm, in bedleveling.cpp line 437, you have a comment:
    //Printer::offsetZ = 0; // we correct this with probe height
In my specific case, with the probe statically mounted alongside the nozzle, I need a configurable "inbetween probes height" variable, is that what this was?

Thanks

repetier, 2016-09-05:
All extruders have a own z offset. Switching between extruders adjusts z height on switch so that should not be part of z probe height.

Thinking about it it might still be wrong. All my printers have special cases where 0 is correct. One is where nozzle is z probe so here I had to not change it. On the other side if z probe is apart from nozzle height we might need to add extruder z offset to measured height to have correct results for each extruder selected. Could you tast if that is the case as you seem to have the required printer for this. I think line 547 would need to be changed

float distance = static_cast<float>(sum) * invAxisStepsPerMM[Z_AXIS] / static_cast<float>(repeat) + EEPROM::zProbeHeight();
or better after it to cover the special case extruder = z probe. We need to add or subtract extruder z offset there if it is now wrong.

Kulitorum, 2016-09-05:
I will try this tomorrow.