repetier/RepetierFirmware #573: MAX6675: wrong results when reading multiple sensors too fast

jsb1, 2016-09-10:
in int16_t read_max6675(uint8_t ss_pin):
Reading two sensors within 230ms will return only the first sensors value for both.

I got trouble in my test code. Not sure if it can happen during normal operation.

int16_t read_max6675(uint8_t ss_pin)
{
    //static millis_t last_max6675_read = 0;
    static int16_t max6675_temp = 0;
 // this is dangerous: reading too fast mixes up sensors!
    //if (HAL::timeInMilliseconds() - last_max6675_read > 230)
    {
...

repetier, 2016-09-11:
Use the version from dev tree. It has a timer for each sensor. The problem is that the sensor does not allow fast reads so we need to wait for next temperature and old version had only one time stored.

closed 2016-09-11 by repetier

jsb1, 2016-09-12:
Sorry, did not check that.My fault :-(