repetier/RepetierFirmware #574: No heating bed on k8200/3drag

lucailvec, 2016-09-12:
The 3Drag can support heating bed but now only the extruder can be used.

The configurator must provide 2 different motherboard:

- 3Drag/Velleman K8200 (experimental)

- 3Drag/Velleman K8200 with heating bed (experimental) 

This is the schema of pinout: 
http://reprap.org/mediawiki/images/9/9f/Schema_base.jpg

repetier, 2016-09-12:
No different boards required. The default has 2 heaters 0 and 1. 1 is by default meant for heated bed, but can also be used for a second extruder. Just check
"Enable heated bed support (HAVE_HEATED_BED)"
to enable bed support and enter the correct heater there.

closed 2016-09-12 by repetier

lucailvec, 2016-09-12:
Ok correct, The board is the same but you can add a button to enable bed on the configurator?

repetier, 2016-09-12:
The button is in tools tab and called "Enable heated bed support (HAVE_HEATED_BED)". It is a simple checkbox.