repetier/RepetierFirmware #575: Dual X Carriage parking posision

Kulitorum, 2016-09-12:
My right hotend homes at x-50 (Behind right is 0)
My left hotend homes at x300

If Hotend 1 is at xpos 100 and I change to head 2, head 0 moves to x0, rather then x-50 which means that hotend 2 can never goto x0, because hotend 1 is here. This causes head crashes.

I belive line 692 in Extruder. cpp:

PrintLine::moveRelativeDistanceInSteps(Extruder::current->xOffset - dualXPos, 0, 0, 0, EXTRUDER_SWITCH_XY_SPEED, true, false);

Should read

PrintLine::moveRelativeDistanceInSteps(dualXPos, 0, 0, 0, EXTRUDER_SWITCH_XY_SPEED, true, false);

Kulitorum

Kulitorum, 2016-09-13:
Hmm, no - thats not the right fix - ideas?

repetier, 2016-09-13:
Maybe a new feature comes in it's way as it worked before. Now we have good destination checks that can prevent moves. So try this instead:
```
	if(Printer::isXHomed() && executeSelect) {
		bool oldDestCheck = Printer::isNoDestinationCheck();
		Printer::setNoDestinationCheck(true);
		PrintLine::moveRelativeDistanceInSteps(Extruder::current->xOffset - dualXPos, 0, 0, 0, EXTRUDER_SWITCH_XY_SPEED, true, false);
		Printer::setNoDestinationCheck(oldDestCheck);
	}

```