repetier/RepetierFirmware #576: Sending "T0" multiple times

Kulitorum, 2016-09-14:
My dualX printer behaves strangely when I select the same head several times, each time the offset_X is applied to the position.

So if my offset_x is -800steps (-10mm) and I send
T0
T0
T0

The head moves to X20 (-10+3x10)

Eeprom dump:

<?xml version="1.0" encoding="utf-8"?><Repetier-Firmware-EEPROM><epr pos="1028" type="0" value="0">Language</epr><epr pos="75" type="2" value="250000">Baudrate</epr><epr pos="129" type="3" value="9265.615">Filament printed</epr><epr pos="125" type="2" value="2182367">Printer active</epr><epr pos="79" type="2" value="0">Max. inactive time</epr><epr pos="83" type="2" value="360000">Stop stepper after inactivity</epr><epr pos="3" type="3" value="80.0000">X-axis steps per mm</epr><epr pos="7" type="3" value="80.0000">Y-axis steps per mm</epr><epr pos="11" type="3" value="1600.0000">Z-axis steps per mm</epr><epr pos="15" type="3" value="500.000">X-axis max. feedrate</epr><epr pos="19" type="3" value="500.000">Y-axis max. feedrate</epr><epr pos="23" type="3" value="10.000">Z-axis max. feedrate</epr><epr pos="27" type="3" value="100.000">X-axis homing feedrate</epr><epr pos="31" type="3" value="100.000">Y-axis homing feedrate</epr><epr pos="35" type="3" value="10.000">Z-axis homing feedrate</epr><epr pos="39" type="3" value="20.000">Max. jerk</epr><epr pos="47" type="3" value="5.000">Max. Z-jerk</epr><epr pos="133" type="3" value="-71.000">X min pos</epr><epr pos="137" type="3" value="0.000">Y min pos</epr><epr pos="141" type="3" value="0.000">Z min pos</epr><epr pos="145" type="3" value="250.000">X max length</epr><epr pos="149" type="3" value="300.000">Y max length</epr><epr pos="153" type="3" value="450.000">Z max length</epr><epr pos="51" type="3" value="2000.000">X-axis acceleration</epr><epr pos="55" type="3" value="2000.000">Y-axis acceleration</epr><epr pos="59" type="3" value="300.000">Z-axis acceleration</epr><epr pos="63" type="3" value="2000.000">X-axis travel acceleration</epr><epr pos="67" type="3" value="2000.000">Y-axis travel acceleration</epr><epr pos="71" type="3" value="300.000">Z-axis travel acceleration</epr><epr pos="1024" type="3" value="0.000">Coating thickness</epr><epr pos="808" type="3" value="-5.900">Z-probe height</epr><epr pos="929" type="3" value="10.000">Max. z-probe - bed dist.</epr><epr pos="812" type="3" value="2.000">Z-probe speed</epr><epr pos="840" type="3" value="150.000">Z-probe x-y-speed</epr><epr pos="800" type="3" value="-22.660">Z-probe offset x</epr><epr pos="804" type="3" value="-9.140">Z-probe offset y</epr><epr pos="816" type="3" value="10.000">Z-probe X1</epr><epr pos="820" type="3" value="20.000">Z-probe Y1</epr><epr pos="824" type="3" value="0.000">Z-probe X2</epr><epr pos="828" type="3" value="280.000">Z-probe Y2</epr><epr pos="832" type="3" value="280.000">Z-probe X3</epr><epr pos="836" type="3" value="305.000">Z-probe Y3</epr><epr pos="1036" type="3" value="0.000">Z-probe bending correction A</epr><epr pos="1040" type="3" value="0.000">Z-probe bending correction B</epr><epr pos="1044" type="3" value="0.000">Z-probe bending correction C</epr><epr pos="880" type="0" value="0">Autolevel active (1/0)</epr><epr pos="106" type="0" value="0">Bed Heat Manager</epr><epr pos="107" type="0" value="255">Bed PID drive max</epr><epr pos="124" type="0" value="80">Bed PID drive min</epr><epr pos="108" type="3" value="196.000">Bed PID P-gain</epr><epr pos="112" type="3" value="33.000">Bed PID I-gain</epr><epr pos="116" type="3" value="290.000">Bed PID D-gain</epr><epr pos="120" type="0" value="255">Bed PID max value</epr><epr pos="1020" type="0" value="0">Enable retraction conversion</epr><epr pos="992" type="3" value="3.000">Retraction length</epr><epr pos="996" type="3" value="13.000">Retraction length extruder switch</epr><epr pos="1000" type="3" value="40.000">Retraction speed</epr><epr pos="1004" type="3" value="0.000">Retraction z-lift</epr><epr pos="1008" type="3" value="0.000">Extra extrusion on undo retract</epr><epr pos="1012" type="3" value="0.000">Extra extrusion on undo switch retract</epr><epr pos="1016" type="3" value="20.000">Retraction undo speed</epr><epr pos="200" type="3" value="95.000">Extr.1 steps per mm</epr><epr pos="204" type="3" value="999.000">Extr.1 max. feedrate</epr><epr pos="208" type="3" value="20.000">Extr.1 start feedrate</epr><epr pos="212" type="3" value="5000.000">Extr.1 acceleration</epr><epr pos="216" type="0" value="3">Extr.1 heat manager</epr><epr pos="217" type="0" value="150">Extr.1 PID drive max</epr><epr pos="245" type="0" value="10">Extr.1 PID drive min</epr><epr pos="218" type="3" value="5.0000">Extr.1 PID P-gain/dead-time</epr><epr pos="222" type="3" value="0.5800">Extr.1 PID I-gain</epr><epr pos="226" type="3" value="19.4100">Extr.1 PID D-gain</epr><epr pos="230" type="0" value="255">Extr.1 PID max value</epr><epr pos="231" type="2" value="0">Extr.1 X-offset</epr><epr pos="235" type="2" value="0">Extr.1 Y-offset</epr><epr pos="290" type="2" value="0">Extr.1 Z-offset</epr><epr pos="239" type="1" value="10">Extr.1 temp. stabilize time</epr><epr pos="250" type="1" value="150">Extr.1 temp. for retraction when heating</epr><epr pos="252" type="1" value="0">Extr.1 distance to retract when heating</epr><epr pos="254" type="0" value="255">Extr.1 extruder cooler speed</epr><epr pos="300" type="3" value="95.000">Extr.2 steps per mm</epr><epr pos="304" type="3" value="999.000">Extr.2 max. feedrate</epr><epr pos="308" type="3" value="20.000">Extr.2 start feedrate</epr><epr pos="312" type="3" value="5000.000">Extr.2 acceleration</epr><epr pos="316" type="0" value="3">Extr.2 heat manager</epr><epr pos="317" type="0" value="150">Extr.2 PID drive max</epr><epr pos="345" type="0" value="10">Extr.2 PID drive min</epr><epr pos="318" type="3" value="5.0000">Extr.2 PID P-gain/dead-time</epr><epr pos="322" type="3" value="0.7800">Extr.2 PID I-gain</epr><epr pos="326" type="3" value="24.0300">Extr.2 PID D-gain</epr><epr pos="330" type="0" value="200">Extr.2 PID max value</epr><epr pos="331" type="2" value="29200">Extr.2 X-offset</epr><epr pos="335" type="2" value="0">Extr.2 Y-offset</epr><epr pos="390" type="2" value="-1200">Extr.2 Z-offset</epr><epr pos="339" type="1" value="10">Extr.2 temp. stabilize time</epr><epr pos="350" type="1" value="150">Extr.2 temp. for retraction when heating</epr><epr pos="352" type="1" value="0">Extr.2 distance to retract when heating</epr><epr pos="354" type="0" value="255">Extr.2 extruder cooler speed</epr></Repetier-Firmware-EEPROM>

Kulitorum, 2016-09-14:
<?xml version="1.0" encoding="utf-8"?><Repetier-Firmware-EEPROM><epr pos="1028" type="0" value="0">Language</epr><epr pos="75" type="2" value="250000">Baudrate</epr><epr pos="129" type="3" value="9265.615">Filament printed</epr><epr pos="125" type="2" value="2182367">Printer active</epr><epr pos="79" type="2" value="0">Max. inactive time</epr><epr pos="83" type="2" value="360000">Stop stepper after inactivity</epr><epr pos="3" type="3" value="80.0000">X-axis steps per mm</epr><epr pos="7" type="3" value="80.0000">Y-axis steps per mm</epr><epr pos="11" type="3" value="1600.0000">Z-axis steps per mm</epr><epr pos="15" type="3" value="500.000">X-axis max. feedrate</epr><epr pos="19" type="3" value="500.000">Y-axis max. feedrate</epr><epr pos="23" type="3" value="10.000">Z-axis max. feedrate</epr><epr pos="27" type="3" value="100.000">X-axis homing feedrate</epr><epr pos="31" type="3" value="100.000">Y-axis homing feedrate</epr><epr pos="35" type="3" value="10.000">Z-axis homing feedrate</epr><epr pos="39" type="3" value="20.000">Max. jerk</epr><epr pos="47" type="3" value="5.000">Max. Z-jerk</epr><epr pos="133" type="3" value="-71.000">X min pos</epr><epr pos="137" type="3" value="0.000">Y min pos</epr><epr pos="141" type="3" value="0.000">Z min pos</epr><epr pos="145" type="3" value="250.000">X max length</epr><epr pos="149" type="3" value="300.000">Y max length</epr><epr pos="153" type="3" value="450.000">Z max length</epr><epr pos="51" type="3" value="2000.000">X-axis acceleration</epr><epr pos="55" type="3" value="2000.000">Y-axis acceleration</epr><epr pos="59" type="3" value="300.000">Z-axis acceleration</epr><epr pos="63" type="3" value="2000.000">X-axis travel acceleration</epr><epr pos="67" type="3" value="2000.000">Y-axis travel acceleration</epr><epr pos="71" type="3" value="300.000">Z-axis travel acceleration</epr><epr pos="1024" type="3" value="0.000">Coating thickness</epr><epr pos="808" type="3" value="-5.900">Z-probe height</epr><epr pos="929" type="3" value="10.000">Max. z-probe - bed dist.</epr><epr pos="812" type="3" value="2.000">Z-probe speed</epr><epr pos="840" type="3" value="150.000">Z-probe x-y-speed</epr><epr pos="800" type="3" value="-22.660">Z-probe offset x</epr><epr pos="804" type="3" value="-9.140">Z-probe offset y</epr><epr pos="816" type="3" value="10.000">Z-probe X1</epr><epr pos="820" type="3" value="20.000">Z-probe Y1</epr><epr pos="824" type="3" value="0.000">Z-probe X2</epr><epr pos="828" type="3" value="280.000">Z-probe Y2</epr><epr pos="832" type="3" value="280.000">Z-probe X3</epr><epr pos="836" type="3" value="305.000">Z-probe Y3</epr><epr pos="1036" type="3" value="0.000">Z-probe bending correction A</epr><epr pos="1040" type="3" value="0.000">Z-probe bending correction B</epr><epr pos="1044" type="3" value="0.000">Z-probe bending correction C</epr><epr pos="880" type="0" value="0">Autolevel active (1/0)</epr><epr pos="106" type="0" value="0">Bed Heat Manager</epr><epr pos="107" type="0" value="255">Bed PID drive max</epr><epr pos="124" type="0" value="80">Bed PID drive min</epr><epr pos="108" type="3" value="196.000">Bed PID P-gain</epr><epr pos="112" type="3" value="33.000">Bed PID I-gain</epr><epr pos="116" type="3" value="290.000">Bed PID D-gain</epr><epr pos="120" type="0" value="255">Bed PID max value</epr><epr pos="1020" type="0" value="0">Enable retraction conversion</epr><epr pos="992" type="3" value="3.000">Retraction length</epr><epr pos="996" type="3" value="13.000">Retraction length extruder switch</epr><epr pos="1000" type="3" value="40.000">Retraction speed</epr><epr pos="1004" type="3" value="0.000">Retraction z-lift</epr><epr pos="1008" type="3" value="0.000">Extra extrusion on undo retract</epr><epr pos="1012" type="3" value="0.000">Extra extrusion on undo switch retract</epr><epr pos="1016" type="3" value="20.000">Retraction undo speed</epr><epr pos="200" type="3" value="95.000">Extr.1 steps per mm</epr><epr pos="204" type="3" value="999.000">Extr.1 max. feedrate</epr><epr pos="208" type="3" value="20.000">Extr.1 start feedrate</epr><epr pos="212" type="3" value="5000.000">Extr.1 acceleration</epr><epr pos="216" type="0" value="3">Extr.1 heat manager</epr><epr pos="217" type="0" value="150">Extr.1 PID drive max</epr><epr pos="245" type="0" value="10">Extr.1 PID drive min</epr><epr pos="218" type="3" value="5.0000">Extr.1 PID P-gain/dead-time</epr><epr pos="222" type="3" value="0.5800">Extr.1 PID I-gain</epr><epr pos="226" type="3" value="19.4100">Extr.1 PID D-gain</epr><epr pos="230" type="0" value="255">Extr.1 PID max value</epr><epr pos="231" type="2" value="0">Extr.1 X-offset</epr><epr pos="235" type="2" value="0">Extr.1 Y-offset</epr><epr pos="290" type="2" value="0">Extr.1 Z-offset</epr><epr pos="239" type="1" value="10">Extr.1 temp. stabilize time</epr><epr pos="250" type="1" value="150">Extr.1 temp. for retraction when heating</epr><epr pos="252" type="1" value="0">Extr.1 distance to retract when heating</epr><epr pos="254" type="0" value="255">Extr.1 extruder cooler speed</epr><epr pos="300" type="3" value="95.000">Extr.2 steps per mm</epr><epr pos="304" type="3" value="999.000">Extr.2 max. feedrate</epr><epr pos="308" type="3" value="20.000">Extr.2 start feedrate</epr><epr pos="312" type="3" value="5000.000">Extr.2 acceleration</epr><epr pos="316" type="0" value="3">Extr.2 heat manager</epr><epr pos="317" type="0" value="150">Extr.2 PID drive max</epr><epr pos="345" type="0" value="10">Extr.2 PID drive min</epr><epr pos="318" type [... cut ...] ><epr pos="331" type="2" value="29200">Extr.2 X-offset</epr><epr pos="335" type="2" value="0">Extr.2 Y-offset</epr><epr pos="390" type="2" value="-1200">Extr.2 Z-offset</epr><epr pos="339" type="1" value="10">Extr.2 temp. stabilize time</epr><epr pos="350" type="1" value="150">Extr.2 temp. for retraction when heating</epr><epr pos="352" type="1" value="0">Extr.2 distance to retract when heating</epr><epr pos="354" type="0" value="255">Extr.2 extruder cooler speed</epr></Repetier-Firmware-EEPROM>

Kulitorum, 2016-09-14:
Rmpffh, I can't paste XML here....

[Repetier-Firmware-2016-09-13_EEprom.zip](https://github.com/repetier/Repetier-Firmware/files/471047/Repetier-Firmware-2016-09-13_EEprom.zip)

repetier, 2016-09-14:
Are you not using latest dev version. There in extruder select you have the following code

```
#if NUM_EXTRUDER > 1 && MIXING_EXTRUDER == 0
    bool executeSelect = false;
    if(extruderId != Extruder::current->id)
    {
        GCode::executeFString(Extruder::current->deselectCommands);
        executeSelect = true;
    }
    Commands::waitUntilEndOfAllMoves();
#endif
    float cx, cy, cz;
    Printer::realPosition(cx, cy, cz);
    float oldfeedrate = Printer::feedrate;
    Extruder::current->extrudePosition = Printer::currentPositionSteps[E_AXIS];
#if DUAL_X_AXIS
	float lastX = Printer::lastCmdPos[X_AXIS];
	// Park current extruder
	int32_t dualXPos = Printer::currentPositionSteps[X_AXIS] - Printer::xMinSteps;
	if(Printer::isXHomed() && executeSelect) {
		bool oldDestCheck = Printer::isNoDestinationCheck();
		Printer::setNoDestinationCheck(true);
		PrintLine::moveRelativeDistanceInSteps(Extruder::current->xOffset - dualXPos, 0, 0, 0, EXTRUDER_SWITCH_XY_SPEED, true, false);
		Printer::setNoDestinationCheck(oldDestCheck);
	}
#endif

```
As you see executeSelect is false if the extruder stays the same and in that case the x move is not executed at all. Maybe adding output debug messages to each move would show better where the move happens.

Kulitorum, 2016-10-10:
This is still messed up. This Gcode is Simplify3d, and after all the init code, it sends a T0 command.

This T0 command makes the head try and move to X-142 or there about:

G90
M82
M106 S0
M140 S60
M190 S60
M104 S230 T0
M104 S190 T1
M109 S230 T0
M109 S190 T1
G21        		;metric values
G90        		;absolute positioning
M82        		;set extruder to absolute mode
M107      		;start with the fan off
G28 X0 Y0  		;move X/Y to min endstops
G28 Z0     		;move Z to min endstops
G92 E0        		;zero the extruded length
G1 F5000
G92 E0
G1 E-5.0000 F1200
G1 Z2.080 F1002
; layer 1, Z = 1.08
T0

Kulitorum, 2016-10-10:
Oh, and I downloaded the whole code yesterday.

Kulitorum, 2016-10-10:
I think its got something to do with my X-min-pos being -71 in the firmware Eeprom config.

repetier, 2016-10-12:
Sorry, but I can not see why that should happen. In Extruder.cpp selectExtruderById you have this

```
#if NUM_EXTRUDER > 1 && MIXING_EXTRUDER == 0
    bool executeSelect = false;
    if(extruderId != Extruder::current->id)
    {
        GCode::executeFString(Extruder::current->deselectCommands);
        executeSelect = true;
    }
    Commands::waitUntilEndOfAllMoves();
#endif
    float cx, cy, cz;
    Printer::realPosition(cx, cy, cz);
    float oldfeedrate = Printer::feedrate;
    Extruder::current->extrudePosition = Printer::currentPositionSteps[E_AXIS];
#if DUAL_X_AXIS
	float lastX = Printer::lastCmdPos[X_AXIS];
	// Park current extruder
	int32_t dualXPos = Printer::currentPositionSteps[X_AXIS] - Printer::xMinSteps;
	if(Printer::isXHomed() && executeSelect) {
		bool oldDestCheck = Printer::isNoDestinationCheck();
		Printer::setNoDestinationCheck(true);
		PrintLine::moveRelativeDistanceInSteps(Extruder::current->xOffset - dualXPos, 0, 0, 0, EXTRUDER_SWITCH_XY_SPEED, true, false);
		Printer::setNoDestinationCheck(oldDestCheck);
	}
#endif
```	

For same extruder executeSelect is false and thus
` if(Printer::isXHomed() && executeSelect) { `
should not execute the move in x direction. Please add some outputs there to see what is really going on. That would be the move I#d expect as T0 does only correct offsets. More over even if it would do so it should be no problem as the offset should be correct, except if this maybe interferes with the lazy positioning trick. Normally a negative offset is what you would need.