repetier/RepetierFirmware #581: Extruder considered jammed on print start

mandrav, 2016-09-27:
Hi,

I have a filament detection switch installed on my printer which should prompt me to change filament, if it runs out.

9 times out of 10, when I start a new print the extruder is considered jammed in the first couple of print seconds. It then starts the change-filament procedure.

By looking at the code, I suspect it's because "jamLastSignal" (in Extruder.h) is not being initialized properly, at least not from what I can see.
I 'm not near the printer atm but will try initializing this variable later today and verify if it's indeed the culprit.

repetier, 2016-09-27:
I never have this problem. But I know that the first layer has often more back pressure so maybe your extruder is slipping more on that layer. With a close error steps count you may then get early signal. Also you need a system that does not send too many signals. A signal every mm or 2 os more then enough and more precise then signaling every 0,1mm. What is your setting here?

mandrav, 2016-09-27:
Thanks for the quick response @repetier !

My "sensor" is a simple switch that changes state when the filament runs out (i.e. the printer is out of filament).
My relevant settings are:
JAM_METHOD="Signal gets high"
JAM_ACTION=""Show jam/out of filament dialog..."

The switch is wired and configured correctly and works as intended.
The only problem I have, hence this issue report, is that when I start a print it will think the switch has changed state (which it hasn't). And that's why I 'm suspecting an uninitialized variable.

Now that I think of it, I have to observe if this happens for the first print after powerup or for any print. This would give an additional clue if it is an uninitialized variable...

Lastly, I forgot to mention that I 'm using the dev branch.

repetier, 2016-09-27:
Ok, that method is kinda untested. We only tested the other method so wrong initialization could be an issue, but then it should happen on enabling printer. They are initialized in printer.cpp at startup using this code:
```
// Initialize jam sensors
#if defined(EXT0_JAM_PIN) && EXT0_JAM_PIN > -1
    SET_INPUT(EXT0_JAM_PIN);
    PULLUP(EXT0_JAM_PIN, EXT0_JAM_PULLUP);
#endif // defined
#if defined(EXT1_JAM_PIN) && EXT1_JAM_PIN > -1
    SET_INPUT(EXT1_JAM_PIN);
    PULLUP(EXT1_JAM_PIN, EXT1_JAM_PULLUP);
#endif // defined
#if defined(EXT2_JAM_PIN) && EXT2_JAM_PIN > -1
    SET_INPUT(EXT2_JAM_PIN);
    PULLUP(EXT2_JAM_PIN, EXT2_JAM_PULLUP);
#endif // defined
#if defined(EXT3_JAM_PIN) && EXT3_JAM_PIN > -1
    SET_INPUT(EXT3_JAM_PIN);
    PULLUP(EXT3_JAM_PIN, EXT3_JAM_PULLUP);
#endif // defined
#if defined(EXT4_JAM_PIN) && EXT4_JAM_PIN > -1
    SET_INPUT(EXT4_JAM_PIN);
    PULLUP(EXT4_JAM_PIN, EXT4_JAM_PULLUP);
#endif // defined
#if defined(EXT5_JAM_PIN) && EXT5_JAM_PIN > -1
    SET_INPUT(EXT5_JAM_PIN);
    PULLUP(EXT5_JAM_PIN, EXT5_JAM_PULLUP);
#endif // defined

```
So  all readings should be ok.

How does it work into the print? Does it trigger there falsely? Could the switch bounce on/off when you need to pull a bit more?

mandrav, 2016-09-27:
The initialization you posted initializes the pins. No problems there. Like I said, it works correctly except when starting up.
Looking in Extruder.h, there's a macro named _TEST_EXTRUDER_JAM.
There, you can see that (for JAM_METHOD!=1) it reads the pin and compares the value against extruder[x].jamLastSignal and, if different than expected, signals the jam.
It's this exact variable (jamLastSignal) that I don't see being initialized anywhere in code. So the first time this comparison is made it may very well fail.

Thanks again @repetier

repetier, 2016-09-27:
Ok, I see the point. Have updated firmware to read signal at startup. So after my above code I added

```
HAL::delayMilliseconds(1);
#if defined(EXT0_JAM_PIN) && EXT0_JAM_PIN > -1
extruder[0].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined
#if defined(EXT1_JAM_PIN) && EXT1_JAM_PIN > -1
extruder[1].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined
#if defined(EXT2_JAM_PIN) && EXT2_JAM_PIN > -1
extruder[2].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined
#if defined(EXT3_JAM_PIN) && EXT3_JAM_PIN > -1
extruder[3].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined
#if defined(EXT4_JAM_PIN) && EXT4_JAM_PIN > -1
extruder[4].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined
#if defined(EXT5_JAM_PIN) && EXT5_JAM_PIN > -1
extruder[5].jamLastSignal = READ(EXT0_JAM_PIN);
#endif // defined

```

That should solve your problem hopefully.

repetier, 2016-09-27:
There is an error in above code. EXT0_JAM_PIN must also be EXT1_JAM_PIN ... - fixed on github/config tool for dev.

mandrav, 2016-09-27:
Great, I will check the fix and report back.

mandrav, 2016-09-27:
Looks like it's working as intended now.
Thanks for the quick fix :)

closed 2016-09-27 by mandrav