repetier/RepetierFirmware #582: Controller resets when doing G28

strips, 2016-09-27:
I have been having some strangeness and now decided to test latest development. 

Did a git pull origin development and am now at ec3c389. Uploaded my latest dev Config to config tool and made some offset changes. Setting E0 as "0" and E1 and Z probe with their offsets relative to E0.

I also did a M502 and M500 but no other changes after this.

When I now connect to printer and do a G28 it finishes Y, X and resets just as it should start Z homing. Maybe I have something wrong in my Config but it should not reset?

[Configuration.h.txt](https://github.com/repetier/Repetier-Firmware/files/495166/Configuration.h.txt)

Octoprint log:
`Recv: wait
Recv: wait
Recv: wait
Recv: start
Send: N0 M110 N0*125
Recv: Info:External Reset
Recv: Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
Recv: zDistortionCorrection:0
Recv: Free RAM:958
Recv: SelectExtruder:0
Recv: FlowMultiply:100
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: M21
Recv: ok 0
Send: M115
Recv: ok 0
Send: M115
Recv: FIRMWARE_NAME:Repetier_1.0.0dev FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
Recv: Printed filament:2.14m Printing time:0 days 1 hours 37 min
Recv: PrinterMode:FFF
Recv: ok 0
Send: M115
Recv: FIRMWARE_NAME:Repetier_1.0.0dev FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
Recv: Printed filament:2.14m Printing time:0 days 1 hours 37 min
Recv: PrinterMode:FFF
Recv: wait
Recv: ok 0
Send: M115
Recv: FIRMWARE_NAME:Repetier_1.0.0dev FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
Recv: Printed filament:2.14m Printing time:0 days 1 hours 37 min
Recv: PrinterMode:FFF
Recv: ok 0
Recv: FIRMWARE_NAME:Repetier_1.0.0dev FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
Recv: Printed filament:2.14m Printing time:0 days 1 hours 37 min
Recv: PrinterMode:FFF
Recv: wait
Recv: wait
Recv: wait
Send: M105
Recv: ok 0
Recv: T:21.52 /0 B:24.51 /0 B@:0 @:0 T0:21.52 /0 @0:0 T1:21.21 /0 @1:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: G28
Recv: ok 0
Recv: busy:processing
Recv: busy:processing
Send: M105
Recv: busy:processing
Recv: start
Recv: Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
Recv: zDistortionCorrection:0
Recv: Free RAM:958
Recv: SelectExtruder:0
Recv: FlowMultiply:100
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Changing monitoring state from 'Operational' to 'Offline'
Connection closed, closing down monitor`

strips, 2016-09-27:
I'm curious about the E1 offsets. Shouldn't they be in mm?
#define EXT1_X_OFFSET 4775
#define EXT1_Y_OFFSET -6710

Just went through my config again. Looks like I entered , as decimal separator like I natively use :(

Building and will retry asap.

strips, 2016-09-27:
Still get reset when doing G28 with:
#define EXT1_X_OFFSET 29.843
#define EXT1_Y_OFFSET -18.64

I had to manually change the values after the config tool wants them to be 4775 and -6710.

repetier, 2016-09-27:
Offset is in steps. Config tool converts it from mm to steps. But that should not be the problem.
What is obvious is that the printer reboots/restarts. Log was missing the reason - normally you see if it was reset/brown out/... Homing z is just moving Z so first test would be moving z without homing. Initally z is at z=0 so moving up should be possible. If it also resets there you know z moves cause reset and you can continue from there. Might be a short causing voltage drop/reboot or just wrong pins for z driver. That is to determine.

strips, 2016-09-27:
Ah, that explains the offset.

Of course I can not 100% rule out brownout but I'm 96% sure it is not as the printer could home fine before upgrading to latest dev from a relative new dev. And I'm able to move the Z axis all the way up and down by knob in display and GCODE/Octoprint.

So my bet is something in the Config combined with the dev branch. Unfortunately I did also some minor offset changes in fw. I could try to go back if I have the older one still.

strips, 2016-09-27:
I flashed an development I compiled 23.09 and now I'm able to home all axis without any reset.

repetier, 2016-10-01:
One thing that sometime happens is with high z steps per mm and high homing speed there is no time to ping watchdog so watchdog triggers. Could that be the issue? Would disabling watchdog make it work? Have tested with latest dev version and homing works normal.

strips, 2016-10-03:
I'll give it a go. As I finally have dual Z max endstops I want to test them :) So I'm loath to go back to Marlin or the stable Repetier branch :)

strips, 2016-10-03:
Just compiled and upladed, only change was disabled watchdog. Still same.

Made a video of homing from display menu: https://photos.google.com/share/AF1QipOMbcAOhMmVyMfoF767l8f96esoa8LdQkKXPWh26KKNvxw6SaYmRsaWifj4_ULrVg/photo/AF1QipNUKalqTPTR1kcdlG0_2QU6q903qmY5pPzQS8mQ?key=WWJWMTUtaXFteGlkb0NUalNCQU5RbEdCRzFBcXV3

There are some strange artefacts ort moves when homing. I'm talking about the extra moves it does. Is this normal?

repetier, 2016-10-04:
I see no unneeded moves. Homing does a lot of moves for retesting and activating extruder offset.

Only untypical is the reset, which I can not say why or exactly when it happens. That would require debugging on your system with some test outputs to find the code part where reset happens. Maybe starting in void Printer::homeZAxis() // Cartesian homing

You would write something like this
Com::printFLN(PSTR("DB1"));HAL::delayMilliseconds(5);

with DB1 changing to DB2 ... at several places. That in host log you will see what point got reached and what not. The delay is to give time for sending output to host before starting crash point.

strips, 2016-10-07:
Thanks, I'll try to locate where that is and start instrumenting the code :)

strips, 2016-10-10:
I got it working . I wasn't getting anywhere and I decided to start from scratch. So not sure about this one. Maybe something in the config or intermittently wrong.

Thanks for your help.

closed 2016-10-11 by repetier