repetier/RepetierFirmware #584: Pica support  (feature request)

dutchrazor, 2016-10-18:
Hello,

I added some code to support the Pica board, made for the Reprap Wilson by MJRice. Should I create a fork in Github, or is it ok to post it here?

Thanks!

Attached the additions:

Added this to pins.h:
`/****************************************************************************************
* MJRice Pica Rev_`B
*
****************************************************************************************/

#if MOTHERBOARD == 183
#define KNOWN_BOARD 1

#define ORIG_X_DIR_PIN 54
#define ORIG_X_STEP_PIN 55
#define ORIG_Y_DIR_PIN 56
#define ORIG_Y_STEP_PIN 57
#define ORIG_Z_DIR_PIN 58
#define ORIG_Z_STEP_PIN 59
#define ORIG_X_ENABLE_PIN 60 
#define ORIG_Y_ENABLE_PIN 61
#define ORIG_Z_ENABLE_PIN 62
#define ORIG_X_MIN_PIN 14
#define ORIG_X_MAX_PIN 15
#define ORIG_Y_MIN_PIN 16
#define ORIG_Y_MAX_PIN 17
#define ORIG_Z_MIN_PIN 23
#define ORIG_Z_MAX_PIN 22

#define ORIG_E0_STEP_PIN         67
#define ORIG_E0_DIR_PIN          24
#define ORIG_E0_ENABLE_PIN       26

#define ORIG_E1_STEP_PIN         68
#define ORIG_E1_DIR_PIN          28
#define ORIG_E1_ENABLE_PIN       27

#define SSR_PIN 6

#define SDPOWER            -1
#define SDSS               53
#define ORIG_SDCARDDETECT       49

#define LED_PIN            -1
#define ORIG_FAN_PIN            11
#define ORIG_FAN2_PIN           12
#define ORIG_PS_ON_PIN          -1

#define HEATER_0_PIN       10
#define HEATER_1_PIN       8
#define HEATER_2_PIN       2
// ANALOG NUMBERING
#define TEMP_0_PIN         9   
#define TEMP_1_PIN         10
#define TEMP_2_PIN         11
#define TEMP_3_PIN         12
#define E0_PINS ORIG_E0_STEP_PIN,ORIG_E0_DIR_PIN,ORIG_E0_ENABLE_PIN,
#define E1_PINS ORIG_E1_STEP_PIN,ORIG_E1_DIR_PIN,ORIG_E1_ENABLE_PIN,

// LCD interface pins
#define LCD_PINS_RS 33
#define LCD_PINS_ENABLE 30
#define LCD_PINS_D4 35
#define LCD_PINS_D5 32
#define LCD_PINS_D6 37
#define LCD_PINS_D7 36
#define BTN_EN1 47
#define BTN_EN2 48
#define BTN_ENC 31 //the click 
#define BLEN_C 2
#define BLEN_B 1
#define BLEN_A 0

#endif

/****************************************************************************************
* MJRice Pica Rev_C
*
****************************************************************************************/

#if MOTHERBOARD == 184
#define KNOWN_BOARD 1

#define ORIG_X_DIR_PIN 54
#define ORIG_X_STEP_PIN 55
#define ORIG_Y_DIR_PIN 56
#define ORIG_Y_STEP_PIN 57
#define ORIG_Z_DIR_PIN 58
#define ORIG_Z_STEP_PIN 59
#define ORIG_X_ENABLE_PIN 60 
#define ORIG_Y_ENABLE_PIN 61
#define ORIG_Z_ENABLE_PIN 62
#define ORIG_X_MIN_PIN 14
#define ORIG_X_MAX_PIN 15
#define ORIG_Y_MIN_PIN 16
#define ORIG_Y_MAX_PIN 17
#define ORIG_Z_MIN_PIN 23
#define ORIG_Z_MAX_PIN 22

#define ORIG_E0_STEP_PIN         67
#define ORIG_E0_DIR_PIN          24
#define ORIG_E0_ENABLE_PIN       26

#define ORIG_E1_STEP_PIN         68
#define ORIG_E1_DIR_PIN          28
#define ORIG_E1_ENABLE_PIN       27

#define SSR_PIN 6

#define SDPOWER            -1
#define SDSS               53
#define ORIG_SDCARDDETECT       49

#define LED_PIN            -1
#define ORIG_FAN_PIN            9
#define ORIG_FAN2_PIN           7
#define ORIG_PS_ON_PIN          -1

#define HEATER_0_PIN       10
#define HEATER_1_PIN       8
#define HEATER_2_PIN       2
// ANALOG NUMBERING
#define TEMP_0_PIN         9   
#define TEMP_1_PIN         10
#define TEMP_2_PIN         11
#define TEMP_3_PIN         12
#define E0_PINS ORIG_E0_STEP_PIN,ORIG_E0_DIR_PIN,ORIG_E0_ENABLE_PIN,
#define E1_PINS ORIG_E1_STEP_PIN,ORIG_E1_DIR_PIN,ORIG_E1_ENABLE_PIN,

// LCD interface pins
#define LCD_PINS_RS 33
#define LCD_PINS_ENABLE 30
#define LCD_PINS_D4 35
#define LCD_PINS_D5 32
#define LCD_PINS_D6 37
#define LCD_PINS_D7 36
#define BTN_EN1 47
#define BTN_EN2 48
#define BTN_ENC 31 //the click 
#define BLEN_C 2
#define BLEN_B 1
#define BLEN_A 0

// these pins are defined in the SD library if building with SD support
#define SCK_PIN          52
#define MISO_PIN         50
#define MOSI_PIN         51
#define MAX6675_SS       53

#endif`

Changed this bit in ui.h:
`#if FEATURE_CONTROLLER == CONTROLLER_SMARTRAMPS || FEATURE_CONTROLLER == CONTROLLER_GADGETS3D_SHIELD || FEATURE_CONTROLLER == CONTROLLER_BAM_DICE_DUE || (FEATURE_CONTROLLER == CONTROLLER_REPRAPDISCOUNT_GLCD && MOTHERBOARD != CONTROLLER_FELIX_DUE)
#undef SDCARDDETECT
#if MOTHERBOARD == 37
#define SDCARDDETECT ORIG_SDCARDDETECT
#elif ( MOTHERBOARD == 183 ) || ( MOTHERBOARD == 184 )

#undef BEEPER_PIN
#define BEEPER_PIN             19
#define UI_DISPLAY_RS_PIN      33
#define UI_DISPLAY_RW_PIN      -1
#define UI_DISPLAY_ENABLE_PIN  30
#define UI_DISPLAY_D0_PIN      -1
#define UI_DISPLAY_D1_PIN      -1
#define UI_DISPLAY_D2_PIN      -1
#define UI_DISPLAY_D3_PIN      -1
#define UI_DISPLAY_D4_PIN      35
#define UI_DISPLAY_D5_PIN      32
#define UI_DISPLAY_D6_PIN      37
#define UI_DISPLAY_D7_PIN      36
#define UI_ENCODER_A           47
#define UI_ENCODER_B           48
#define UI_ENCODER_CLICK       31
#define UI_RESET_PIN           -1
#define SDCARDDETECT 49

#else
#define SDCARDDETECT 49
#endif
#undef SDCARDDETECTINVERTED
#define SDCARDDETECTINVERTED 0
#undef SDSUPPORT
#define SDSUPPORT 1
#endif`

repetier, 2016-10-19:
No fork required. Will add it to development later this day.

repetier, 2016-10-19:
I have problems with your ui.h changes. I don't think you indicated the right place so I put them where I think they belong to. Hope it works that way.

Online config tool now also contains the new boards.

closed 2016-10-19 by repetier

dutchrazor, 2016-11-25:
Thanks! I tested your changes and it works perfectly.

My ui.h changes were a bit wonky indeed.