repetier/RepetierFirmware #586: step_units calculation issue

eeerfan, 2016-10-19:
Hi dears

i'm try to make my own 3d printer.before complete the project and testing,i think i must config my step motors by changing step_units,when i calculate step per unit(mm) i reach this number (19.6850) but this number dos not config my stepper motor correctly,but when i product this number to 10(19.6850*10=196.85) and add to configuration all of things are ok and machine works correctly.
my calculations:

StepsPerTurn = 400 // Most stepper motors have 0.9 Degree per step
BeltDistance = 2.032mm // Distance between repeating grooves. GT2 belt = 2.032mm
GroovesInPulley = 10 // 8-10 are common values---> FOR ME IS 10
Substepping = 1//for a full step needed
StepsPerMM = StepsPerTurn*Substepping/(BeltDistance**GroovesInPulley)
           = 400*1/(2.032*10) = 19.6850

my question is that,why the number must product to 10 in full step controlls?

repetier, 2016-10-19:
Your motor has 1.8 degress like most printers
Your pulley has 8 grooves
Your substepping is 1/16 like most have
=> 197.0443 steps per mm.

This is for me the most likely reason.

rcarlyle, 2016-10-19:
What stepper driver are you using and what microstepping is it doing? Your calculation is missing microstepping. Full-stepping is highly inadvisable because it has severe resonance issues. At minimum, you should use quarter stepping for smooth motion. 1/8 to 1/32 is better. 

GT2 should be 2.000mm, but MXL is 2.032mm. Are you using GT2 or MXL? Most printers now use GT2. 

Most steppers are 1.8 degree / 200 steps/rev, please confirm if you have 200 or 400.

8 grooves is extremely unusual for GT2, the minimum officially supported by Gates is 12 grooves. Please confirm if 8 is correct. (If so, you will have greatly reduced belt life.)

eeerfan, 2016-10-19:
my stepper motor is 0.9 degree based on that my step per revolution is 400
and i must use full step control  for project(my teacher want that)
and i use GT2  but 2.000 and 2.032 dose not make lot of changes.

rcarlyle, 2016-10-19:
Ok, your correct steps/mm for full-stepping is 20 steps per mm. That's 400 steps/rev for the motor, no microstepping, and a 10 tooth GT2 2mm pulley. 400/(10*2)=20. If that value isn't giving you the proper travel, your stepper driver is probably configured wrong.

eeerfan, 2016-10-19:
i use drv 8825 for step motor driving.whats mean'stepper driver configured wrong'?
and how i can config them correctly?

rcarlyle, 2016-10-19:
Check the microstep setting selection jumpers on your control board, or if there are no jumpers, check the board schematic to see what is wired up. Compare to the 8825 datasheet to see what you're actually getting.

rcarlyle, 2016-10-19:
This really isn't a firmware issue, you might try somewhere like the RepRap forums or 3D Printing Google Plus community for troubleshooting help.

closed 2016-10-22 by eeerfan