repetier/RepetierFirmware #588: Add- BLTouch Capabilities

dcwalmsley, 2016-10-24:
I built a Cartesian 3D Printer and procured a BLTouch sensor for Z-Axis.  
http://www.ANTCLABS.com has specifications and information which will be of use for developers.

Currently, I'm toying around with Marlin 1.1.0 RC7 and they have BLTouch incorporated in their Firmware.  But there is things with Repetier which is driving my towards using this firmware over Marlin.  Only setback for me is not knowing how to use BLTouch with Repetier. 

What can I do to use it with your firmware?

repetier, 2016-10-24:
What makes the BLtouch different from other sensors so that it does not work? With our scripts for activation and deactivation we normally cover all z probes.

dcwalmsley, 2016-10-25:
@repetier,  I'm not sure I can answer your question.
1.  I have very little experience with your firmware.
2. BLTouch is different than capacitance and proximity sensors. It uses a touch probe that is retracted when not in use.  It uses two wire bundles, a 2-pin for Zin-Min connection and the other (3-pin) for power and pin extension/retraction.  

This URL provides much more information. https://plus.google.com/113792662927481823969

Guess what I need to know is how to configure Repetier to use this sensor?

repetier, 2016-10-25:
Have you tried this:
https://forum.repetier.com/discussion/2477/bltouch-configuration-on-makergear-m2-with-rambohttp://

I guess if you check the z probe select/deselect commands you know how to do it.

Kulitorum, 2017-01-17:
For reference, I made the BLTOUCH work with repetier yesterday.
Attached my config.h that makes it work.
You connect the black/white wires to your zmin endstop GND/SIGNAL
And the servo connects to EXP3 pin 2,4,6
2: +5v (Red)
4:GND (Orange)
6:Signal (brown)

[image]
[image]
[image]
[Configuration.zip](https://github.com/repetier/Repetier-Firmware/files/712524/Configuration.zip)

shaiss, 2017-03-17:
@Kulitorum I have a rumba and BLtouch as well.  Thank you for sharing!  Have you gotten this up and running?  Could you share an updated config file?  Thank you!

ArchGalileu, 2017-03-28:
@Kulitorum please share ;)

Best regards
ArchGalileu

shaiss, 2017-03-28:
@archgalileu.  my config is at https://1drv.ms/u/s!ApzOdVx9w7haifw1Tc8G1fmHvbuZ6Q and it works.  at the moment I home to zmax and then g32 s2 to calculate the z height.   

in the next couple of days I'm going to change to home to zmin and try that

ArchGalileu, 2017-03-28:
@shaiss you use the bltouch to Z home?

shaiss, 2017-03-28:
@archgalileu I will be not yet.  I home to zmax and use the bltouch to calculate the z height.  that's repetier's standard.   recently in the dev branch there's been alot of work for using the probe as the zmin and homing to zmin.  hence I plan on trying that soon as I nail down my probe offset and some other dual extruder settings

webliya, 2017-04-20:
After trying bltouch to work for 2 weeks I decided to opt it out. Now I purchased an 8mm inductive sensor and will try it.

I had several problems with voltage, auto bed leveling, nozzle height. It works with Marlin but with Repetier I believe still needs some development.

shaiss, 2017-04-20:
I got it to work only after countless 10s of hours.  I don't know if it was
worth the headache. I don't think it's repetier's fault that the bltouch
doesn't work. It has some quirks we need to sort out.  I'll post my setup
soon as I'm ironing out the wrinkles.

webliya, 2017-04-21:
Will be waiting for your setup. I still want to use it because it seems simple and small.

hergonoway, 2017-05-21:
A friend of mine use it on Marlin, damn this sensor is so precise. It should (with its generic friend the 3dtouch) be implemented in repetier that's for sure.

gc9n, 2018-01-09:
Any news about it??

repetier, 2018-01-09:
As fas as I know it works with repetier if the right configuration is used. After all it is just a sensor plus servo and we have commands to control servos.

apphead, 2018-01-17:
Could you please provide a sample _right configuration_ and add to the z-probe documentation?

repetier, 2018-01-18:
Don't have a BLtouch so can not say what is good. But if someone can say what exactly is required to get it work I can add that to documentation.

repetier, 2018-01-18:
From @shaiss config it seems

```
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 2
#define Z_PROBE_REPETITIONS 1#define Z_PROBE_HEIGHT 1.175 //was 0.6
#define Z_PROBE_START_SCRIPT "M340 P0 S700"
#define Z_PROBE_FINISHED_SCRIPT "M340 P0 S1500"
````
is the only special part.

apphead, 2018-01-18:
Hmm, what about HOMING_ORDER HOME_ORDER_XYZ....
See https://github.com/repetier/Repetier-Firmware/issues/623#issuecomment-304052780

repetier, 2018-01-18:
That is a good idea for all z probes and is needed if you can not activate probe at home x,y.

BoaImpressao3D, 2020-11-09:
> What makes the BLtouch different from other sensors so that it does not work? With our scripts for activation and deactivation we normally cover all z probes.

The difference is tha BLTouch must be unalarmed befor use. So at Z_PROBE_START_SCRIPT must have: "M340 P0 S2190\nM340 P0 S650"
The rest is the same.