repetier/RepetierFirmware #589: step_mm failure

eeerfan, 2016-10-24:
Hi dears
I try to calculate my step_mm in this way:
StepsPerTurn = 400 //my stepper motors are 0.9 degree 
BeltDistance = 2.0mm // GT2 belt 
GroovesInPulley = 10 // FOR ME IS 10
Substepping = 16//for a 1/16 step needed
StepsPerMM = StepsPerTurnSubstepping/(BeltDistanceGroovesInPulley)
= 40016/(2.0*10) =320

but when i add this number for step per mm my printer dosent work correctly.
what can i do???

rcarlyle, 2016-10-24:
This is not enough information for us to help you troubleshoot, and it pretty clearly isn't a Repetier firmware issue. You should post pictures of your hardware and a description of your troubleshooting steps taken so far on a general help site like the RepRap forum.

repetier, 2016-10-25:
Did you change it in eeprom or configuration.h? EEPROM would be the right place.

closed 2016-10-29 by eeerfan