repetier/RepetierFirmware #592: while before } error

shaiss, 2016-11-03:
I wanted to try the dev version, but when comping, in the Arduino ide on line 2369, I get an error "expected while before } token".

`#endif
                    for(y = 0; y < UI_ROWS; y++)
                        printRow(y, &cache[y][off[y]], NULL, UI_COLS);
#if UI_DISPLAY_TYPE == DISPLAY_U8G
                    }
                }
#endif
#if UI_DISPLAY_TYPE == DISPLAY_U8G
        } ///<<<Error is marked as here in the IDE
        while( u8g_NextPage(&u8g) );  //end picture loop
#endif
#endif
    Printer::toggleAnimation();
}`

not sure what is causing this as this section seems to be the same as .92

shaiss, 2016-11-03:
[image]

repetier, 2016-11-03:
Only happened when UI_HEAD was not defined. Have uploaded the corrected version. Without UI_HEAD the result is the same with your fix.

shaiss, 2016-11-03:
Was I supposed to define ui_head somewhere in the configurator tool?   I just used my config.h from .92 and changed some values for v1 like jam detection going high and dual z mins. 

I'll download this new version and try compiling later.

shaiss, 2016-11-03:
just confirmed that the fw compiles fine now and looked through your last commit.  I'll close this issue out since it resolves the issue.  Will upload to the printer later and confirm.  ty for the hard work!

closed 2016-11-03 by shaiss

repetier, 2016-11-03:
With graphic display you can define UI_HEAD to have a fixed status bar. If you don't you get the old menu style. So not required to define it.