repetier/RepetierFirmware #596: Cooldown menu action disables board fan while Z-axis driver still active

frazierjason, 2016-11-12:
Repro'd on v0.92.9, MendelMax 1.0, RUMBA board, RepRapDiscount Full Graphics LCD:

1) Compile firmware with PREVENT_Z_STEPPER_DISABLE_ON_TIMEOUT enabled, and with a board cooling fan configured
2) Power up printer with a board fan connected
3) Home all axes using the LCD menu, and verify the board cooling fan starts spinning
4) Using the LCD menu, go to Quick Settings > select Cooldown

Actual:
The board cooling fan and X/Y motors disengage, but the Z screws are still on.  This could cause the Z stepper driver to overheat.

Expected:
The board cooling fan should have remained on for as long as the Z screws remain powered.

repetier, 2016-11-12:
Agreed. Would also happen if you manually disable heaters. Github version is updated, online tool will follow when I'm through with some others mods.

frazierjason, 2016-11-13:
Thanks for fixing this so fast, I'll await the next online tool update.  I've put in five bucks on Paypal as a token of appreciation.

How can I tell when you've updated the online tool with a newer build if the firmware revision's major/minor/micro didn't change?

repetier, 2016-11-14:
You can not see it, but I have updated it just now.

closed 2016-11-14 by repetier