repetier/RepetierFirmware #597: Request:  Park Z steppers in a non-fractional motor position before disabling driver

frazierjason, 2016-11-12:
I'd like to suggest a feature, if it isn't in already.  Upon gracefully powering down a given double-motor axis (Z in my case), it would be ideal to first tell the driver(s) to move both motors towards the safest direction (away from the build platform) until it reaches the next full angle step position (no half-step, for example).  This could help prevent each motor from drifting unpredictably as it powers down.

When the Z axis is disabled in a dual-Z motor printer, or when the printer loses power or is rebooted, each of the stepper motors will lose their microstep position and align to whatever full step position is most attractive at the time (quite literally).  This could be further influenced by differences in how each motor experiences the power loss.  It's possible that each stepper motor occasionally becomes attracted to the next full step in the opposite direction from each other.  Over time, this could cause the Z stepper motors to drift apart, positionally speaking, and cause the X axis to no longer be tram with the build platform since one side is higher than the other.

Like many Cartesian printers, I have a Mendelmax with an X axis raised by two Z axis 1/8" threaded rod, and Z motors that are not coupled to each other (besides sharing a driver wired in parallel).  I also have a WeldingRodBot (Cobblebot) with a 15^2 inch gantry raised with two Z screws.  I eventually tied the WeldingRodBot Z screws together with a timing belt loop to eliminate drift (to be fair, this was partially due to gantry weight).  I've just rebuilt the MendelMax with 8mm lead screws that operate more smoothly, but also with greater lift per angle.  I'm curious to know if anyone has studied this issue.

rcarlyle, 2016-11-12:
My general observation is that printers with multiple non-synchronized screws are most often build with single-start screws (eg all-thread or 2mm pitch lead screw) so the thread angle is self-locking and there's minimal issue with drift on powering down. "Fast" screws like 8mm lead screws should be belt-synchronized because the motor's detent torque simply isn't enough to keep it from back-driving.

frazierjason, 2016-11-12:
The 8mm lead screw on my WeldingRodBot did have issues easily back-driving due to the large gantry weight, but I haven't seen any issues so far on the MendelMax -- its X axis and carriage is just not that heavy, I don't use manual force on it (Bowden tubes), and there is just an ever slight bit of binding along each side's screw/rod pair to require a little bit of torque to turn the screws.  Detent force is enough to stay parked.  But when powering up, I can clearly hear the motors jumping into a full-step position, suggesting they were left in whatever micro-step they drifted to on power down.

rcarlyle, 2016-11-12:
Back-driving issue aside, detent torque pulls the rotor to the nearest of the four full-step positions on power-down, but the power-on position of the stepper driver is a specific one of the four half-step position. So jumping on power-up is basically a given. The only question is how far and what direction it jumps. In order to control that, the firmware would have to track its absolute step phase at all times relative to the power-on position. (Skipped steps wouldn't matter,  but dropped microsteps due to endstop hits or whatever would.) I'm not familiar enough with the code to say how difficult that is, but it's probably a non-trivial amount of work, to address what is fundamentally a hardware shortcoming.

repetier, 2016-11-12:
For parking this is more something you would put into your end gcode so you stop where you want.

I understand the problem, but not so sure if that would help at all. First problem would be that we do not know the substepping we are in as we do not count it normally. That would be a big source of cpu usage for higher speeds.  Also as @rcarlyle already noted it is more a hardware shortcoming for which there are several solutions. dev version supports 2 z axis endstops as the easy solution to that problem. Then homing would correct it automatically to a identical level. The not so easy solution would be a coupling belt. Both would work equally well.

frazierjason, 2016-11-13:
Fixing hardware issues in software is actually an awesome approach when it's cheap and uncomplicated to do.  Unfortunately as pointed out here, the tradeoff is too great for the benefit this time around.  Closing issue, thanks for all the thoughtful comments.

closed 2016-11-13 by frazierjason