repetier/RepetierFirmware #599: G31 doesn't relay trigger info

5stringmonster, 2016-11-22:
I'm using an npn inductive sensor. High, then low when triggered. I've hooked it all up to a 9v independently with the voltage divider proven to work by others. The sensor works just fine, and when it's all connected to an mks gen 1.4 board, the sense connection from sensor reads 3.3v not triggered, then drops low when triggered. When I run a g31 it reads high no matter what. I've tried all different settings. Normal-open, normal-closed, electronic normal-closed and open. Inverted, non-inverted, and still I can't get g31 to read the sensor correctly. Using website, configuration 0.92.9. I did verify sensor connections with a multi-meter.

repetier, 2016-11-22:
Did you set it only for z probe or also as z min? In latter case both settings must be identical (no pullup). Also note that one of the z probe settings is defined opposite to z min - think it was invert.

designermx2, 2016-11-22:
5stringmonster, Do you have the model and schematich used? I was trying a simple switch and the repetier don´t work with single switch. It would be great to have a "no filament" option and only work with one simple endstop button.

5stringmonster, 2016-11-22:
I'm using a LJ18A3-8-Z/BX NO switch. The schematic for the voltage divider is the one used in tom's guide for inductive sensors on youtube. 15k ohm from sense to z-min endstop sense, 10k ohm from sense to ground. Leaves me with 3.3v and switches properly when tested on a 9v battery, though the voltages read are at 2.3v using that battery. On the board itself it gives me a proper 3.3v. I'm not using the pullups for either zmin endstop, nor for z-min probe. The inverting signal seems to work though. When I invert it, g31 gives a low signal no matter what, when it's non-inverted it gives a high signal no matter what. Also, I have the sensor positive plugged in to a 12v pin. Well enough for the sensor to use.